VR-25/acc #115: Deamon stops running after some time

Dark-Noir, 2021-09-10:
Hey,

i am new to ACC and using ACCA for the config. I already tried to update to the latest ACC version (v2021.9.5 (202109050)) without any change in the behavior. 

Phone: 
OnePlus 3T @ LOS 17.1 (Android 10)

I got logs from console menu, but not sure if i should upload it publicly.

Switch123456789, 2021-09-13:
I've got the same issue since a few days, I was using acc for years now on my i9300 (LOS18.1) without any issues ever. Now it just broke. When I start the daemon, it works fine but after 5 seconds it turns off and the phone starts charging again. I will upload logs if needed.

VR-25, 2021-09-13:
[acc_v2021.9.13_(202109130).zip](https://github.com/VR-25/acc/files/7154160/acc_v2021.9.13_.202109130.zip)
Try that build and report back.

Switch123456789, 2021-09-13:
Oh boi, it's working with this build =) thank you sir.

Dark-Noir, 2021-09-13:
> 
> 
> [acc_v2021.9.13_(202109130).zip](https://github.com/VR-25/acc/files/7154160/acc_v2021.9.13_.202109130.zip)
> Try that build and report back.

Flashed it via the ACC menu via termux. Didn't work, also crashes after few seconds after plugged in.

VR-25, 2021-09-13:
> > [acc_v2021.9.13_(202109130).zip](https://github.com/VR-25/acc/files/7154160/acc_v2021.9.13_.202109130.zip)
> > Try that build and report back.
> 
> Flashed it via the ACC menu via termux. This build don't work, also crashes after few seconds after plugged in.

It seems to be an issue on your end.
A log archive (`acc -le`) would be great.

Dark-Noir, 2021-09-13:
> It seems to be an issue on your end.
> A log archive (`acc -le`) would be great.

Sure thing, can you send me your email, so i can send you the archive?

VR-25, 2021-09-13:
Unlike my email, those logs are very bland and impersonal.
They contain zero personal information.
If you still don't want to share them here, you can upload the archive to a private folder in my [dropbox](https://www.dropbox.com/request/RmzrlZ3eEN0qGPi1RuL6).

Dark-Noir, 2021-09-13:
> 
> 
> Unlike my email, those logs are very bland and impersonal.
> They contain zero personal information.
> If you still don't want to share them here, you can upload the archive to a private folder in my [dropbox](https://www.dropbox.com/request/RmzrlZ3eEN0qGPi1RuL6).
Okay here is the archive:
[acc-logs-OnePlus3T.tar 1.gz](https://github.com/VR-25/acc/files/7157141/acc-logs-OnePlus3T.tar.1.gz)

VR-25, 2021-09-13:
ALLOW_HVDCP3=1
BATTERY_4P4V_PRESENT=1
**BATTERY_CHARGING_ENABLED=0**
BATTERY_HEALTH=79
BATTERY_INFO=-1
BATTERY_INFO_ID=0
BATTERY_TYPE=OP_3400mAH
CAPACITY=100
CAPACITY_RAW=9612
CHARGE_COUNTER=2966000
CHARGE_FULL=2966000
CHARGE_FULL_DESIGN=2966000
CHARGE_NOW=0
CHARGE_NOW=5131
CHARGE_NOW_ERROR=812
CHARGE_NOW_RAW=2850112
CHARGE_TYPE=N/A
CHARGING_ENABLED=1
CHG_PROTECT_STATUS=0
CONSTANT_CHARGE_CURRENT_MAX=1900000
CURRENT_MAX=1800000
**CURRENT_NOW=0**
CYCLE_COUNT=0
CYCLE_COUNT_ID=1
DP_DM=0
ENABLE_JEITA_DETECTION=0
ESR_COUNT=0
FASTCHG_STARTING=0
FASTCHG_STATUS=1
FLASH_ACTIVE=0
FLASH_CURRENT_MAX=2426413
FLASH_TRIGGER=0
HEALTH=Good
HI_POWER=0
IGNORE_FALSE_NEGATIVE_ISENSE=1
INPUT_CURRENT_LIMITED=0
INPUT_CURRENT_MAX=1800000
INPUT_CURRENT_NOW=-22
INPUT_CURRENT_SETTLED=0
MAX_PULSE_ALLOWED=-22
NG_COUNT=0
PRESENT=1
RERUN_AICL=0
RESISTANCE=208984
RESISTANCE_ID=10000
RESTRICTED_CHARGING=0
SAFETY_TIMER_ENABLED=1
SOC_REPORTING_READY=1
**STATUS=Full**
SYSTEM_TEMP_LEVEL=0
TECHNOLOGY=Li-ion
TEMP=279
TEMP_COOL=100
TEMP_WARM=400
UPDATE_NOW=0
VOLTAGE_MAX=4360
VOLTAGE_MAX_DESIGN=4370000
VOLTAGE_MIN=0
VOLTAGE_NOW=4337000
VOLTAGE_OCV=4280065

CURRENT_NOW=0 Amps
VOLTAGE_NOW=4.337 Volts
POWER_NOW=0 Watts

Are you sure this wasn't the state of charge when acc daemon stopped?
If the battery is already full, acc cannot tell whether its _charging stop_ requests are honored.
It assumes that all charging switches fail to honor the requests.
After the _disable_ request is sent, the expected status is "Not charging" or "Discharging".
It reads "Full" instead, which does not make sense in acc's context.

If this is what really happened, let your battery discharge a bit, run `accd` to restart the daemon, plug* the charger and see if charging stops.

* Test with USB power or a regular (non OEM) adapter first.

Dark-Noir, 2021-09-13:
When i use a normal charger, it takes some seconds to disable charging (it goes off, than on and back off again). When i connect than to a fast charger, it stays off. But when i reboot my phone and than plug in directly the fast charger, acc stops working.
[acc-logs-OnePlus3T.tar.gz](https://github.com/VR-25/acc/files/7157378/acc-logs-OnePlus3T.tar.gz)

Dark-Noir, 2021-09-13:
When i used the trick mentioned above to stop charging with a normal charger and plugin the fast charger again. It seems acc loses the fight against my OP fast charger. When my battery drops below the threshold and acc allows charging again, it tries to fight against the charger (charging goes on and off for a minute or so), but ultimately loses and the daemon stops working again.
[acc-logs-OnePlus3T.tar.gz](https://github.com/VR-25/acc/files/7157519/acc-logs-OnePlus3T.tar.gz)

Dark-Noir, 2021-09-14:
> 
> 
> This is a kernel issue, as described in _zip/README > Troubleshooting > Charging Switch_.
> There are several workarounds in the documentation.
> Read the aforementioned section and _Tips > Idle Mode and Alternatives_.
> You may prefer _zip/README.html_, since it can be viewed on a regular web browser.

Okay, thanks for the info, but i not really understand that section. Should acc not automatically use the oneplus charging switch? How i can check which switch settings are in use?. There is a gui for that in ACCA, but i also don't get that 😅 

I think, i just buy a new battery when it dies. My phone is now nearly 5 years old and the battery is still okay for me, its somewhat easy to swap, so why not. Thanks for the help anyway 👍

VR-25, 2021-09-14:
> Okay, thanks for the info, but i not really understand that section. Should acc not automatically use the oneplus charging switch? How i can check which switch settings are in use?. There is a gui [...]

Here's what's going on:
ACC expects "Discharging" or "Not charging" status after _asking_ the switches to disable charging.
When all switches _refuse to comply_, the daemon stops with exit code `7`.
This is a kernel issue.
All charging switches are failing to respond.

You can disregard the reading and follow my troubleshooting guide.
While the charger is plugged, what's the output of the `acc -t` command? (Run on terminal as root.)

Dark-Noir, 2021-09-14:
Just "...won't work" for all.
[image]

VR-25, 2021-09-14:
Ok, with the charger still plugged, **paste** and run the following on a root terminal - and share the output:

`accd. > /dev/null; cd /sys/class/power_supply/; echo 0 > battery/battery_charging_enabled; sleep 5; acca -i status=\|charge_type=; echo 0 > battery/charging_enabled; sleep 5; acca -i status=\|charge_type=; echo 0 > dc/charging_enabled; sleep 5; acca -i status=\|charge_type=`

er1z, 2021-09-14:
Unfortunately, the latest compilation doesn't work on Pixel 4a 4G. I'm using Acca, daemon starts but in a matter of seconds terminates. Sent you logs to the Dropbox (sunfish).

VR-25, 2021-09-14:
> Unfortunately, the latest compilation doesn't work on Pixel 4a 4G. I'm using Acca, daemon starts but in a matter of seconds terminates. Sent you logs to the Dropbox (sunfish).

Got it.

VR-25, 2021-09-14:
@er1z, try [acc_v2021.9.14.1_(202109141).zip](https://github.com/VR-25/acc/files/7161359/acc_v2021.9.14.1_.202109141.zip).

Dark-Noir, 2021-09-14:
Here is the output of two executions, the first one when limit is not reached, the second one at limit:
[image]
[image]

er1z, 2021-09-14:
@VR-25 seems to be not restarting anymore. But once I connect a charger and decide „charge once without…” I can notice „discharging (taper)”. Is any setting responsible for this?

VR-25, 2021-09-14:
> @VR-25 seems to be not restarting anymore. But once I connect a charger and decide „charge once without…” I can notice „discharging (taper)”. Is any setting responsible for this?

This may be the same issue described in my previous comment.

VR-25, 2021-09-15:
[acc_v2021.9.15.1_(202109151).zip](https://github.com/VR-25/acc/files/7172169/acc_v2021.9.15.1_.202109151.zip)

Dark-Noir, 2021-09-15:
> 
> 
> [acc_v2021.9.15.1_(202109151).zip](https://github.com/VR-25/acc/files/7172169/acc_v2021.9.15.1_.202109151.zip)

Here is the new output of acc -t and the other command that you posted:
[image]
[image]

er1z, 2021-09-15:
> [acc_v2021.9.15.1_(202109151).zip](https://github.com/VR-25/acc/files/7172169/acc_v2021.9.15.1_.202109151.zip)

Doesn't fix my issue, unfortunately. Do you need fresh logs?

VR-25, 2021-09-16:
https://github.com/VR-25/acc/files/7176032/acc_v2021.9.16.1_.202109161.zip

er1z, 2021-09-16:
Now doesn't discharge at plug, can control current and suppose it'll stop once reaches a max threshold.

Thank you very much, this is a module I always been looking for!

PS. Is it possible to support via Patreon once? Or is it only possible by a subscription?

VR-25, 2021-09-16:
> Now doesn't discharge at plug, can control current and suppose it'll stop once reaches a max threshold.
> 
> Thank you very much, this is a module I always been looking for!
> 
> PS. Is it possible to support via Patreon once? Or is it only possible by a subscription?
> 
> update: once the threshold was reached, status changed to „charging: taper” and daemon deads. Once restarted, nothing changes and charging doesn't stop.

You're welcome.

Patreon only accepts [recurrent](https://support.patreon.com/hc/en-us/articles/204606215-Can-I-make-a-one-time-payment-) payments.
Liberapay has this _symbolic_ donation that is close enough:
~$0,01/week, ~ $0,04/month ~$0,52/year
I'm only currently getting one-time donations through Paypal.
Suggestions in regards to other services are appreciated.

Could you share fresh logs?
We're getting closer to a good workaround.

er1z, 2021-09-16:
Check the Dropbox.

VR-25, 2021-09-16:
[acc_v2021.9.16.2_(202109162).zip](https://github.com/VR-25/acc/files/7176595/acc_v2021.9.16.2_.202109162.zip)

VR-25, 2021-09-16:
Refer to issue #106 for newer builds.

Colo-Thor, 2021-09-17:
```
OnePlus8T:/ # accd,

(i) accd is not running

9|OnePlus8T:/ # acc -t

(!) Charger must be plugged to continue...
/system/bin/acc[580]: curr_dropped: parameter not set
1|OnePlus8T:/ #
[acc-logs-OnePlus8T.tar.gz](https://github.com/VR-25/acc/files/7183135/acc-logs-OnePlus8T.tar.gz)

```

VR-25, 2021-09-17:
[acc_v2021.9.17_(202109170).zip](https://github.com/VR-25/acc/files/7188476/acc_v2021.9.17_.202109170.zip)

Colo-Thor, 2021-09-18:
> [acc_v2021.9.17_(202109170).zip](https://github.com/VR-25/acc/files/7188476/acc_v2021.9.17_.202109170.zip)

It work for me

Dark-Noir, 2021-09-18:
> 
> 
> [acc_v2021.9.17_(202109170).zip](https://github.com/VR-25/acc/files/7188476/acc_v2021.9.17_.202109170.zip)

Still not work here, "acc -t" reports "won't work" for all switches and the daemon also stops after few seconds when charging fast if the set limit is reached.

VR-25, 2021-09-19:
> > [acc_v2021.9.17_(202109170).zip](https://github.com/VR-25/acc/files/7188476/acc_v2021.9.17_.202109170.zip)
> 
> Still not work here, "acc -t" reports "won't work" for all switches and the daemon also stops after few seconds when charging fast if the set limit is reached.

This is a kernel issue.
The daemon stops precisely because all switches fail to respond.

closed 2021-09-19 by VR-25