VR-25/acc #116: Old Xperia tablet crashes / reboots upon sime acc activities

elem089, 2021-09-19:
Hi,
just found issue #107 which reminded me of some similar problems on one of my older Xperia devices (Xperia Tablet Z, running LineageOS 16).
I'm running acc 2021-9-5 on it, and it does its job. But stopping, starting and restarting the daemon from terminal the tablet crashes and reboots.
Testing for charging switches and other activities though don't cause any trouble.
I haven't tried to use some of the "--set" activities as I usually edit the config.txt by hand.
I already tried some alternative kernel but behaviour is exactly the same.
Strange thing is that this problem started sometime around Feb or March 2021 after I had updated the ROM. I remember I reported this through the telegram channel, but since I no longer use telegram I don't know what happened.

Anyways, after reading the mentioned issue I ran `acc -le` to get that log, and here it is:
[acc-logs-pollux_windy.tar.gz](https://github.com/VR-25/acc/files/7192182/acc-logs-pollux_windy.tar.gz)

Thanks and best regards!

VR-25, 2021-09-19:
Do `acc -d`, `acc -e` and `acc -t` also trigger reboots?

elem089, 2021-09-19:
`acc -d` yes
`acc -e` yes, if charger is plugged, no if on battery
`acc -t` same as `acc -e`. But strange: if charger is not plugged, and if I plug it in after the warning message appeared then it does its job telling me about the supported switches, and doesn't crash/reboot``

VR-25, 2021-09-19:
Upgrade to the latest version (just released) and share a new log archive.
This is most likely a charging switch issue.

elem089, 2021-09-19:
I did, and apperently the crashes are gone! This is cool, had expected my device being too old to cope. 
I guess you no longer need the log now, or do you?

Have a nice evening!

elem089, 2021-09-19:
Right after flashing the problem appeared to be gone, no more reboots. Then I left the device lying for 2 hours and forgot to unplug the charger.
Next time I looked at it it was sitting constantly at 76% which I found odd because max charge is set to 80%. So I tried to force a restart calling accd. And bam! Next reboot. Tried various of the other options but Reboots are now again triggered everytime.
so here's the next log, I'm afraid
Thanks still and again
[acc-logs-pollux_windy.tar.gz](https://github.com/VR-25/acc/files/7192759/acc-logs-pollux_windy.tar.gz)

elem089, 2021-09-21:
quick updated: last night upgraded to latest version (2021-09-20), and again thought that now the reboots were gone. whatever I did in terminal: device stayed up and running.
This morning I dared to give it another try, this time with the charger plugged in (USB-charging from my laptop, to be exact). Tried acc -t >> reboot
Back in system, still plugged in via USB - tried simple accd >> reboot
Back in system, with USB unplugged - again tried simple accd >> reboot.
My conclusion: after flashing a (new) version the device appears to gain some kind of stability, but ass soon as the charger was plugged in once the system becomes instable again. Does that make any sense?
a) not sure whether you need yet another log, but here it is ;)
b) of course, and again: in my opinion this is a minor issue. I'll be happy to provide additional info, do some more testing if you want me to, but on the other hand I personally could easily live with the situation as it is now
Regards
[acc-logs-pollux_windy.tar.gz](https://github.com/VR-25/acc/files/7201027/acc-logs-pollux_windy.tar.gz)
!

VR-25, 2021-09-23:
`echo 1 > /sys/module/pm8921_charger/parameters/disabled (disable charging)`
`echo 0 > /sys/module/pm8921_charger/parameters/disabled (enable charging)`

See if any of those commands trigger reboots when the charger is plugged and when it ins't.

VR-25, 2021-09-23:
See also if `acc -t /sys/module/pm8921_bms/parameters/battery_removed 0 1` reports success.

elem089, 2021-09-24:
> 
> 
> `echo 1 > /sys/module/pm8921_charger/parameters/disabled` (disable charging)
> `echo 0 > /sys/module/pm8921_charger/parameters/disabled` (enable charging)
> 
> See if any of those commands trigger reboots when the charger is plugged and when it ins't.
 no apparent reaction, neither plugged in nor offline

elem089, 2021-09-24:
> 
> 
> See also if `acc -t /sys/module/pm8921_bms/parameters/battery_removed 0 1` reports success.

forced reboot if charger is plugged in

VR-25, 2021-09-24:
> > `echo 1 > /sys/module/pm8921_charger/parameters/disabled` (disable charging)
> > `echo 0 > /sys/module/pm8921_charger/parameters/disabled` (enable charging)
> > See if any of those commands trigger reboots when the charger is plugged and when it ins't.
> 
> no apparent reaction, neither plugged in nor offline

But does charging actually pause/resume?

VR-25, 2022-02-03:
The upcoming release will auto detect and blacklist unwritable control files and those that trigger unexpected reboots.

closed 2022-02-03 by VR-25