VR-25/acc #127: Doesn't work on Pixel4

PenguinDOOM, 2021-11-06:
OS: Android12
Kernel: Kirisakura Kernel

It's set to stop at 60% and go into idle mode, but it keeps charging without stopping.
When I look at the ACCA monitoring, it seems to stop and charge repeatedly.
[acc-logs-flame.tar.gz](https://github.com/VR-25/acc/files/7490597/acc-logs-flame.tar.gz)

VR-25, 2021-11-06:
This is a kernel issue introduced by Android 12.

[acc-p.txt](https://github.com/VR-25/acc/files/7490609/acc-p.txt)
That file is in the log archive and it holds the output of acc -p - which is a list of potential charging switches not present in acc's database.
Run `acc -t /sdcard/Download/acc-p.txt` to test all and report back.

PenguinDOOM, 2021-11-06:
The only switch that supported idle mode was [usb/otg_fastroleswap 0 1].
[acc-t.txt](https://github.com/VR-25/acc/files/7490902/acc-t.txt)

VR-25, 2021-11-06:
I'll add the two working switches to the database.
You can set the one that supports idle mode: `acc -s s="usb/otg_fastroleswap 0 1 --"`

PenguinDOOM, 2021-11-07:
I set it to `usb/otg_fastroleswap` but it keeps charging.
[acc-logs-flame.tar.gz](https://github.com/VR-25/acc/files/7492431/acc-logs-flame.tar.gz)

VR-25, 2021-11-08:
How slow does it charge? What's the current?

PenguinDOOM, 2021-11-08:
About 10% per hour.

VR-25, 2021-11-08:
See if you can determine the average current with `acc -w1 curr` or `acc -i curr`
I'm working on a new agorithm to detect idle mode based on current threshods - and this data is fundamental.

PenguinDOOM, 2021-11-08:
[image]

VR-25, 2021-12-12:
Try [this](https://github.com/VR-25/acc/files/7696614/acc_v2021.12.11_202112110.zip) build.

closed 2021-12-25 by PenguinDOOM