VR-25/acc #128: Support A205xx battery idle

cledsupper, 2021-11-07:
Device: SM-A205G
OS: Android 11 (R) / OneUI 3.1

ACC doesn't seem to support 'battery idle' feature of Galaxy A20.

I even tried adding the sysfs switch file manually on AccA (ACC frontend). While testing, AccA says it works, but ACC can't detect battery stop charging and automatically change to the default charging switch.

I believe this happened because 'status' continue to report that the battery is "Charging", while 'current_now' varies between -10 ~ 10 mA once the charge switch is set to 0.

The file below is part of a program I made to get an equivalent "conservation mode" feature (available on Ideapad 320) to work on my phone. It contains all relevant information.

[battery.sh](https://github.com/cledsupper/batservice-a20/blob/magisk-module/lib/battery.sh)

VR-25, 2021-11-08:
Working on it.

VR-25, 2021-12-03:
[acc_v2021.12.3_202112030.zip](https://github.com/VR-25/acc/files/7651938/acc_v2021.12.3_202112030.zip)
Let me know whether this build gets the job done.

cledsupper, 2021-12-04:
I used the 'acc' command to select the switch and also the charge levels.
But ACC doesn't even seem to touch the 'hmt_ta_charge' file.

cledsupper, 2021-12-04:
[acc-logs-a20.tar.gz](https://github.com/VR-25/acc/files/7654372/acc-logs-a20.tar.gz)

cledsupper, 2021-12-04:
Sorry if my actions on this issue make some noise.
I noticed through some thorough research that the logs generated by ACC expose a lot of sensitive data from my device that is unnecessary for service development.
[acc-logs-a20-directorscut.tar.gz](https://github.com/VR-25/acc/files/7654416/acc-logs-a20-directorscut.tar.gz)

VR-25, 2021-12-04:
Idle mode support will get better.

I'll also work on minimizing the amount of information that is logged and exclude potentially sensitive bits.

Thanks for the input.

VR-25, 2021-12-05:
[acc_v2021.12.5_202112050.zip](https://github.com/VR-25/acc/files/7655798/acc_v2021.12.5_202112050.zip)

cledsupper, 2021-12-06:
I don't know if this is an app issue, but AccA is showing the battery is discharging and current is a negative value.

[image]

I'll be posting the logs soon.

cledsupper, 2021-12-06:
[acc-logs-a20-directorscut.zip](https://github.com/VR-25/acc/files/7662935/acc-logs-a20-directorscut.zip)

VR-25, 2021-12-06:
> [acc_v2021.12.5_202112050.zip](https://github.com/VR-25/acc/files/7655798/acc_v2021.12.5_202112050.zip)

Have you tried this build?

cledsupper, 2021-12-06:
> > [acc_v2021.12.5_202112050.zip](https://github.com/VR-25/acc/files/7655798/acc_v2021.12.5_202112050.zip)
> 
> Have you tried this build?

Yes. I uploaded that logs and image right after testing the build, and it didn't work.

VR-25, 2021-12-07:
> Yes. I uploaded that logs and image right after testing the build, and it didn't work.

That's weird, considering that the logs contain v2021.12.3, dmesg.txt and a prop file with sensitive information.
The new version does not have any of that.

cledsupper, 2021-12-07:
> > Yes. I uploaded that logs and image right after testing the build, and it didn't work.
> 
> That's weird, considering that the logs contain v2021.12.3, dmesg.txt and a prop file with sensitive information. The new version does not have any of that.

Maybe AccA had installed an old build again. I'm going to test it one more time.

cledsupper, 2021-12-07:
Mistake was mine. When selecting the zip file, I didn't paid attention for the file selector opened in another directory

cledsupper, 2021-12-07:
ACC seems to read the charge status inverted. But I can't opine on this as I haven't studied the source code.

[acc-logs-a20.tar.gz](https://github.com/VR-25/acc/files/7665773/acc-logs-a20.tar.gz)

VR-25, 2021-12-07:
> ACC seems to read the charge status inverted. But I can't opine on this as I haven't studied the source code.
> 
> [acc-logs-a20.tar.gz](https://github.com/VR-25/acc/files/7665773/acc-logs-a20.tar.gz)
> 

I'll have a look at the logs later.
Meanwhile, see if you can pinpoint issues in `acc/batt-interface.sh`.

VR-25, 2021-12-11:
[acc_v2021.12.11_202112110.zip](https://github.com/VR-25/acc/files/7696614/acc_v2021.12.11_202112110.zip)
Try this.

cledsupper, 2021-12-12:
Great!

But now ACC crashes, at least according to AccA app.

[acc-logs-a20.tar.gz](https://github.com/VR-25/acc/files/7699142/acc-logs-a20.tar.gz)

VR-25, 2021-12-12:
[acc_v2021.12.12_202112120.zip](https://github.com/VR-25/acc/files/7699211/acc_v2021.12.12_202112120.zip)

cledsupper, 2021-12-17:
[acc-logs-a20.tar.gz](https://github.com/VR-25/acc/files/7734883/acc-logs-a20.tar.gz)

VR-25, 2021-12-17:
How about [this](https://forum.xda-developers.com/attachments/acc_v2021-12-15-1_202112151-zip.5485065/) build?

cledsupper, 2021-12-19:
Now it's working fine!

closed 2022-02-03 by VR-25