VR-25/acc #130: Charging current issue

pritamdstan, 2021-11-12:
) just wanted to limit the temparature. But whenever i make some changes, it stops charging at a certain level around 75. And then i stop the acc service. It resumes charging but charges very slow at like 200 mA. I unplug and plug charger again but same current. I even stop and restart service but still same condition. It doesn't charge above 200 or 300 mA. Please advice
[Logs](https://drive.google.com/file/d/1-eTY-9J2sz1C-nke4B3THvyTXr3YXn7W/view?usp=drivesdk) `

VR-25, 2021-12-12:
Try [this](https://github.com/VR-25/acc/files/7696614/acc_v2021.12.11_202112110.zip) build.

closed 2022-02-03 by VR-25

pritamdstan, 2022-03-28:
> Try [this](https://github.com/VR-25/acc/files/7696614/acc_v2021.12.11_202112110.zip) build.

Thanks it works on latest builds.