VR-25/acc #134: Xiaomi Renoir issue

Hogehopo, 2022-01-06:
I'm using Xiaomi's renoir.
The name of CustomRom is FlokoROM.
ACC does not work with this ROM.
Charging does not stop even if the specified charging value is reached.
Attach the Log.
Please tell me what to do.
thank you.
[acc-logs-renoir.tar.gz](https://github.com/VR-25/acc/files/7819832/acc-logs-renoir.tar.gz)

Hogehopo, 2022-01-17:
Please tell me　(T_T)

VR-25, 2022-01-17:
You're not using the latest version, my friend.
This is a requirement for reporting issues.
Download and install the [latest](https://github.com/VR-25/acc/releases/latest) flashable zip.
See if the issue persists.
A new release will be available in a few days.

Hogehopo, 2022-01-18:
> You're not using the latest version, my friend. This is a requirement for reporting issues. Download and install the [latest](https://github.com/VR-25/acc/releases/latest) flashable zip. See if the issue persists.

Thank you for your reply。I tried running the latest version。However, charging does not come beyond the set upper temperature
[acc-logs-renoir0118.tar.gz](https://github.com/VR-25/acc/files/7887466/acc-logs-renoir0118.tar.gz)
。I will attach Log, so please give me some advice

Hogehopo, 2022-01-18:
The question is, does acc support SD780G?

VR-25, 2022-01-23:
The actual question is, does your kernel support acc? :)
A new release will be up soon.

closed 2022-02-03 by VR-25