VR-25/acc #140: Feature request: updateJson attribute for magisk

v1993, 2022-01-29:
Now that magisk's repository is no longer maintained, modules are expected to define `updateJson` attribute in order to allow updating from withing magisk application. It would be really nice for ACC to utilize this new functionality.

VR-25, 2022-01-29:
It's already implemented.
You'll see it in the next release, coming up soon.

v1993, 2022-01-29:
Nice, looking forward to it! Additionally, can you consider submitting to [new module repository](https://github.com/Magisk-Modules-Alt-Repo/submission)? It's maintained by enthusiasts rather than Magisk creator and still in early stages for now, but people are clearly interested in it.

VR-25, 2022-01-29:
Sure!

closed 2022-02-03 by VR-25