VR-25/acc #143: The IDLE charging mode stopped working with the newest release

morfikov, 2022-02-04:
I recently updated ACC to the version 2022-02-03, and I noticed that the IDLE mode stopped working, i.e. when the charging limit is reached, there's no power drained from the charger. I flashed (via magisk) the older version 2021.12.20 and it looks like the IDLE mode works again. 

The device is MTK MT6769T (Xiaomi Redmi 9) and it was using the following charging switch:

```
(i) [/proc/mtk_battery_cmd/current_cmd 0::0 0::1 /proc/mtk_battery_cmd/en_power_path 1 0] works
- battIdleMode=true
```
I noticed that with the newer ACC version, the `battIdleMode` often reports `false` here (not always, but often). But even when it reports `true`, the IDLE mode doesn't work. I also tried other switches, but none of them worked either.

VR-25, 2022-02-06:
[acc_2022-02-06_10.59.37.zip](https://github.com/VR-25/acc/files/8009900/acc_2022-02-06_10.59.37.zip)
- Consider STATUS=Discharging if current drops by mA >= 100 after calling disable_charging();
- New presets for charging_switch=<mA|mV>;
- Use idleThreshold=95 (mA) for mtk devices and 15 for the rest.

Try it.

morfikov, 2022-02-06:
I installed the version you've provided, but there's no change, i.e. the IDLE mode doesn't work and the switch in question still returns `battIdleMode=false` . 

Should I make some changes to the config file?

VR-25, 2022-02-06:
This means `battIdleMode=true` was a false positive that resulted from miscalculations in previous versions.

Notice that there are two types of idle mode:
1. Native (straightforward);
2. Emulated (all about charging status detection; I implemented it and constantly work to improve the algorithms).

Two is what you're dealing with.
I also have an mtk device, and the switch you're using was also inconsistent when it comes to idle mode support – because the emulated idle mode algorithms were in early stages of development.
Now, it clearly does not support it.

Stay tuned for updates.
Meanwhile, you may want to [explore new charging switches](https://github.com/VR-25/acc#finding-additionalpotential-charging-switches-quickly) and [idle mode alternatives](https://github.com/VR-25/acc#idle-mode-and-alternatives).

closed 2022-02-06 by VR-25

morfikov, 2022-02-06:
I see, thanks for the info.