VR-25/acc #151: Is there a way to make the battery charge to its physical max capacity?

Hikandy, 2022-05-25:
Basically I replaced the 3000mah battery in phone with a 5000mah one but its only getting charged till 3000mah (tested the estimated battery capacity via apps like accubattery, 3C battery monitor etc), so is there a way to let the battery charge to its max capacity instead of the one defined by the system by modifying some voltage or current parameters?

VR-25, 2022-06-02:
It's possible, if your kernel allows editing /sys/class/power_supply/*/charge_full. See if that interface exists and whether you can change the value after granting write permission to the root user.

VR-25, 2022-06-26:
Any update on this?
Have you tried the above procedure or need additional help?

Hikandy, 2022-06-26:
Unfortunately, it doesn't seem to work. Even if i edit the files, it simply
goes back to the original value after exiting the file. Is it possible to
adjust (idk much) the values of voltage or current to override the preset
values to charge the battery to max extent?

VR-25, 2022-06-26:
Generally, current and voltage do not affect this.
Share a log .tgz archive generated (`acc -le`) by the latest pre-release.
I'll see if there's something else for adjusting the maximum battery capacity.

closed 2022-06-28 by VR-25