VR-25/acc #152: ACC won't run (command inaccessible)

natewu, 2022-05-29:
I updated my version to the latest and ACCA suddenly stopped working. So I tried to downgrade back to the original version and it still won't work. Then I reinstalled both. Still not working, so I check if I can access acc from the terminal(termux) and it says command not found. I enter SU and still same thing. Then I check ACCA logs and it says ```"/system/bin/sh: <stdin>[25]: acc: inaccessible or not found"```

Not sure what's happening?

vonguyenkhang, 2022-06-05:
v2022.6.4 doesn't work.
AccA is not responding.
accd. command doesn't work.
Trying to revert to previous version but unable install (installation process is not responding, plugging in charger didn't help)
Now phone is unable to charge at all.
Trying to uninstall Magisk module.
Please, have better quality check before posting a stable release.

VR-25, 2022-06-06:
Wipe AccA's data.