VR-25/acc #153: v2022.6.4 doesn't work

vonguyenkhang, 2022-06-05:
v2022.6.4 doesn't work.
AccA is not responding.
accd. command doesn't work.
Trying to revert to previous version but unable install (installation process is not responding, plugging in charger didn't help)
Now phone is unable to charge at all.
Update:
Recovered by uninstalling acc Magisk module and reverted to 2022.2.22.1
Please, have better quality check before posting a stable release.
Xiaomi Redmi Note 9 Pro Global (joyeuse) V12.5.5.0.RJZMIXM (Android 11)

IMgoRt, 2022-06-05:
I also find that ACC doesn't work - I only use command line not AccA
OnePlus 8 A12 EU C20

zhugexinmo, 2022-06-05:
The latest version does not reach the set battery level when charging
Xiaomi 11    MUI13.0.8   Android12

VR-25, 2022-06-06:
Logs, please...

AccA hasn't been updated in a while. Wiping its data is a common solution to these problems.

Regarding acc, see if removing `/data/adb/vr25/acc-data/` helps.

Quality control is taken care of by a group of testers (myself included). No version is released as stable if it doesn't work for each testers in that group.

The problem is that we don't have access to every single device out there.

IMgoRt, 2022-06-06:
Reinstalled, same behaviour, logs attached OnePlus 8
[acc-logs-OnePlus8.zip](https://github.com/VR-25/acc/files/8842504/acc-logs-OnePlus8.zip)

VR-25, 2022-06-06:
> Reinstalled, same behaviour, logs attached OnePlus 8 [acc-logs-OnePlus8.zip](https://github.com/VR-25/acc/files/8842504/acc-logs-OnePlus8.zip)

Run `acc -t` and share `/sdcard/Download/acc-t_output.txt`.

IMgoRt, 2022-06-06:
> Attached [acc-t_output.txt](https://github.com/VR-25/acc/files/8845406/acc-t_output.txt)

Now seems to be working, I'll revert tomorrow if issues occur

vonguyenkhang, 2022-06-06:
1. Sorry for the rude comments above, I was on edge recently due to high stress at work. 1st post is edited.
2. I've reinstalled acc v2022.6.4 using Magisk, then restarted the phone (Xiaomi Redmi Note 9 Pro Global (joyeuse) V12.5.5.0.RJZMIXM (Android 11))
3. Logs are attached (generated from acc v2022.6.4 menu) [acc-logs-joyeuse.zip](https://github.com/VR-25/acc/files/8845646/acc-logs-joyeuse.zip)
4. config.txt is attached (from acc v2022.6.4) [config.txt](https://github.com/VR-25/acc/files/8845643/config.txt)
5. ```accd.``` command takes 27sec to return. 
6. I know acc and AccA are independent works, but without AccA it's difficult for laymen as me to see what's the status (charging/not charging/idle), so I hope AccA can continue to work as it used to.
7. AccA app cache & data is cleared. In the attached screenshot, phone is plugged to charger, system battery icon show charging, but AccA show "Discharging" (expected: charing), and the charging current is higher than set (config: 130mA, if working properly it takes a long time to charge, but in reality the battery % increases very fast - kind of quick charge - please let me know if you need me to install a 3rd party app to monitor current) [image]
8. Charging switch "battery/charging_enabled  1 0" doesn't work anymore (it used to work with acc v2022.2.22.1) [acc-t_output.txt](https://github.com/VR-25/acc/files/8845822/acc-t_output.txt)
9. Retest charging switch after 10sec: now "battery/charging_enabled  1 0" is working [acc-t_output (2).txt](https://github.com/VR-25/acc/files/8845857/acc-t_output.2.txt)
10. ```accd.``` command now takes 5sec to return. 
11. AccA app now shows all info correctly without clearing cache & data.
12. Another set of logs and config.txt (acc v2022.6.4) when everything are working: [config (2).txt](https://github.com/VR-25/acc/files/8845936/config.2.txt) [acc-logs-joyeuse (2).zip](https://github.com/VR-25/acc/files/8845935/acc-logs-joyeuse.2.zip)

IMgoRt, 2022-06-06:
> 1. Sorry for the rude comments above, I was on edge recently due to high stress at work. 1st post is edited.
> 2. I've reinstalled acc v2022.6.4 using Magisk, then restarted the phone (Xiaomi Redmi Note 9 Pro Global (joyeuse) V12.5.5.0.RJZMIXM (Android 11))
> 3. Logs are attached (generated from acc v2022.6.4 menu) [acc-logs-joyeuse.zip](https://github.com/VR-25/acc/files/8845646/acc-logs-joyeuse.zip)
> 4. config.txt is attached (from acc v2022.6.4) [config.txt](https://github.com/VR-25/acc/files/8845643/config.txt)
> 5. `accd.` command takes 27sec to return.
> 6. I know acc and AccA are independent works, but without AccA it's difficult for laymen as me to see what's the status (charging/not charging/idle), so I hope AccA can continue to work as it used to.
> 7. AccA app cache & data is cleared. In the attached screenshot, phone is plugged to charger, system battery icon show charging, but AccA show "Discharging" (expected: charing), and the charging current is higher than set (config: 130mA, if working properly it takes a long time to charge, but in reality the battery % increases very fast - kind of quick charge - please let me know if you need me to install a 3rd party app to monitor current) [image]
> 8. Charging switch "battery/charging_enabled  1 0" doesn't work anymore (it used to work with acc v2022.2.22.1) [acc-t_output.txt](https://github.com/VR-25/acc/files/8845822/acc-t_output.txt)
> 9. Retest charging switch after 10sec: now "battery/charging_enabled  1 0" is working [acc-t_output (2).txt](https://github.com/VR-25/acc/files/8845857/acc-t_output.2.txt)
> 10. `accd.` command now takes 5sec to return.
> 11. AccA app now shows all info correctly without clearing cache & data.
> 12. Another set of logs and config.txt (acc v2022.6.4) when everything are working: [config (2).txt](https://github.com/VR-25/acc/files/8845936/config.2.txt) [acc-logs-joyeuse (2).zip](https://github.com/VR-25/acc/files/8845935/acc-logs-joyeuse.2.zip)
> 
> I've renamed the topic to reflect what I observed. Please rename or close the issue if needed.

I use AccuBattery to monitor charge current

VR-25, 2022-06-06:
>     1. Sorry for the rude comments above, I was on edge recently due to high stress at work. 1st post is edited.

I usually blame our broken societies for this sort of bug (reaction).
You fixed the "bug" like a boss, kinda rare to see nowadays.
Thank you.

>     4. config.txt is attached (from acc v2022.6.4) [config.txt](https://github.com/VR-25/acc/files/8845643/config.txt)

The log archive includes config.txt and pretty much any other relevant data.
 
>     5. `accd.` command takes 27sec to return.

Most likely cause: the charger was unplugged.
Unfortunately, some devices require the charger to be plugged for defaults to be restored.
Restoring default charging settings is part of the daemon stop process.
Those defaults are also restored on reboot, since RAM is cleared.

Upgrading/downgrading acc also implies stopping the daemon.
Having the charger unplugged at that stage sometimes causes minor issues.
Such problems can be solved by re-initializing the daemon (`accd -i`; not the same as restarting it).

>     6. I know acc and AccA are independent works, but without AccA it's difficult for laymen as me to see what's the status (charging/not charging/idle), so I hope AccA can continue to work as it used to.

I'll talk to AccA devs soon.
There's another front-end called [AccSettings](https://github.com/CrazyBoyFeng/AccSettings/releases/). It's not as pretty and user-friendly as AccA, but it's lighter and much more stable.

> 7. AccA app cache & data is cleared. In the attached screenshot, phone is plugged to charger, system battery icon show charging, but AccA show "Discharging" (expected: charing), and the charging [...]
Those are related to `#5`.

To sum up, common (and even some more complex) issues can be solved by running `acc --init` (or -i), rebooting, and/or clearing acc's\* (and front-end's) data. We have most of these issues because Android is a fragmented mess. If only it were an actual GNU/Linux distribution... a distant dream, it seems.

\* `/data/adb/vr25/acc-data/`

zhugexinmo, 2022-06-07:
I installed the script via Magisk.
I directly modify the configuration in config.txt to achieve the required settings.
capacity=(2 70 75 80 false false)
Only 1 of the 4 charges reaches 80%, and the other 3 charge to 76%-77%
How are the logs sent to you?

zhugexinmo, 2022-06-07:
I don't know if I'm doing it right.
Open acc.sh and type acc-le.
I took a screenshot because I won't export the file.
The uploaded name is my github name, and the email address is filled in at will.

vonguyenkhang, 2022-06-07:
the strange behaviour occurs again today:
steps:
1. Charging current doesn't match with setting: AccA status: -5xx mA, setting: 300mA
2. Restart phone
3. ```accd,```  -> running. Charging current still doesn't match with setting as step 1.
4. ```acc -i``` Charging current still doesn't match with setting as step 1.
5. ```acc -t``` (first time): "battery/charging_enabled 1 0" Switch doesn't work. Charging current still doesn't match with setting as step 1.
6. ```acc -t``` (first time): "battery/charging_enabled 1 0" Switch works, idleMode=true (expected). Charging current matched with setting: AccA: 3xx mA.
Full output of the above commands: [cmd-output.txt](https://github.com/VR-25/acc/files/8854337/cmd-output.txt)
Logs: [acc-logs-joyeuse.zip](https://github.com/VR-25/acc/files/8854348/acc-logs-joyeuse.zip)
I haven't cleared /data/adb/vr25/acc-data/ yet.

vonguyenkhang, 2022-06-08:
@VR-25 I've uninstalled acc, cleared acc data ```rm -r /data/adb/vr25/acc-data/``` then reinstalled acc v2022.6.4
I also removed AccA and used [AccSettings](https://github.com/CrazyBoyFeng/AccSettings/releases/) v2022.6.7
Now everything are good.

 @IMgoRt @zhugexinmo you should also try uninstalling acc & AccA, then clear acc data as @VR-25 suggested.

zhugexinmo, 2022-06-08:
> @VR-25我卸载了 acc，清除了 acc 数据，`rm -r /data/adb/vr25/acc-data/`然后重新安装了 acc v2022.6.4 我还删除了 AccA 并使用[了 AccSettings](https://github.com/CrazyBoyFeng/AccSettings/releases/) v2022.6.7 现在一切都很好。
> 
> @IMgoRt @zhugexinmo您还应该尝试卸载 acc & AccA，然后将 acc 数据清除为@VR-25建议。

I uninstalled through magisk, the data will be deleted automatically, I confirmed it.
I uninstalled and reinstalled multiple times to no avail.

rnd2822, 2022-06-08:
Same problem here with a OnePlus 8 stock A11. Reinstalled / deleted data multiple times without success.

IMgoRt, 2022-06-08:
Likewise, it's unstable - sometimes working sometimes not
The Feb version is rock stable
1+8 A12 EU c20

vonguyenkhang, 2022-06-09:
> @VR-25 I've uninstalled acc, cleared acc data `rm -r /data/adb/vr25/acc-data/` then reinstalled acc v2022.6.4 I also removed AccA and used [...]

@IMgoRt @zhugexinmo @VR-25 sorry for the haste conclusion, I've just encounter the issue again.
I still don't know the steps to trigger the issue, and what's the exact nature of the issue, but here's what I observed:
1. Uninstall, cleanup, reinstall acc, reboot, redo setting from scratch using AccSettings (no import)
2. Everything works normally even after unplugging/replugging charger multiple times, starting/stopping accd multiple times. Battery is charged to the correct level set.
3. When I wake up and use my phone, battery level drops and charing is triggered, Android shows battery charging icon, but charging current is not as what has been set in acc, and AccSettings app shows rapid switching between charging-idle-discharging, and battery level still drops. After restarting accd, it's back to idle and everything seems normal again.Video: 
https://user-images.githubusercontent.com/15910168/172743534-9df4104d-08e3-4685-82ce-208d4ed91386.mp4

VR-25, 2022-06-10:
[acc_v2022.6.10-dev_202206100_1431.zip](https://github.com/VR-25/acc/files/8879239/acc_v2022.6.10-dev_202206100_1431.zip)
Try this build.

IMgoRt, 2022-06-11:
> [acc_v2022.6.10-dev_202206100_1431.zip](https://github.com/VR-25/acc/files/8879239/acc_v2022.6.10-dev_202206100_1431.zip) Try this build.

Been using this for 24+ hrs now - no issues to report

Thanks

vonguyenkhang, 2022-06-12:
> [acc_v2022.6.10-dev_202206100_1431.zip](https://github.com/VR-25/acc/files/8879239/acc_v2022.6.10-dev_202206100_1431.zip) 
Try this build.

Still not working for me. Steps:
1. Uninstall acc
2. Reboot
3. ```rm -r /data/adb/vr25/acc-data/```
4. ```rm -r /dev/.vr25/acc```
5. Clear data of "AccSettings"
6. Install acc v2022.6.10-dev
7. Reboot
8. ```acc -i```
9. ```acc -t``` (2 times: result is consistent now)
10. ```acc -le``` → [acc-logs-joyeuse 2022.6.10-dev.zip](https://github.com/VR-25/acc/files/8885115/acc-logs-joyeuse.2022.6.10-dev.zip)
11. ```acc -ss``` → select "battery/charging_enabled 1 0" (this setting supports both charging switch and idle mode)
12. ```accd,```
13. ```accd.``` → this command takes 20sec to return (in v2022.2.22.1: ```accd.``` takes 3sec to return)
14. ```accd```
15. ```accd,```  → AccSettings app : Battery status switches between "Charging"/"Idle"/"Discharging", even after reboot
16. Uninstall acc
17. Reboot
18. clean up & resinstall acc v2022.2.22.1, reboot
19. config acc v2022.2.22.1 via adb shell
20. start AccSettings → acc v2022.2.22.1 is auto updated to v2022.6.4 !!!
21. uninstall & clean up again, reinstall acc v2022.2.22.1, reboot
22. reconfig acc v2022.2.22.1 → all good

[image]
---

**Additional question**: how to set the charging current limit without using either AccA app or AccSettings app?
AccA app auto generated this config ```maxChargingCurrent=(130 /sys/class/qcom-battery/restrict_cur::130000::1000000 battery/constant_charge_current_max::130000::5900000 dc/current_max::130000::100000 main/constant_charge_current_max::130000::5900000 main/current_max::130000::500000 pc_port/current_max::130000::500000 usb/current_max::130000::500000)```, but AccSettings generated a different config (I forgot to backup).
Is there any documentation on how to generate that setting?

---

**Additional rant:** AccSettings seems inferior to AccA: unable to toggle "Battery Idle", unable to switch profiles, incorrectly reinstall acc when acc is already installed (?), etc. I think AccSettings app is more compatible with acc just because AccSettings doesn't have as many features as AccA app.

IMgoRt, 2022-06-12:
> Still not working for me. Steps:
>
> 1. Uninstall acc
> 2. Reboot
> 3. `rm -r /data/adb/vr25/acc-data/`
> 4. `rm -r /dev/.vr25/acc`
> 5. Clear data of "AccSettings"
> 6. Install acc v2022.6.10-dev
> 7. Reboot
> 8. [...]

Yep, having reported everything ok yesterday, it would charge last night
Back to Feb/22

VR-25, 2022-06-12:
[acc_v2022.6.12-dev_202206120_1403.zip](https://github.com/VR-25/acc/files/8885900/acc_v2022.6.12-dev_202206120_1403.zip)
- (wip) Always check real batt status before comparing current
- 1076997 Fix "charging keeps stopping at min < capacity < max"

Please report ASAP whether this stops charging at the desired capacity.

vonguyenkhang, 2022-06-18:
> [acc_v2022.6.12-dev_202206120_1403.zip](https://github.com/VR-25/acc/files/8885900/acc_v2022.6.12-dev_202206120_1403.zip)
> 
> * (wip) Always check real batt status before comparing current (@IMgoRt @vonguyenkhang)
> * [1076997](https://github.com/VR-25/acc/commit/107699713cae7d11c9cad93e0d11b213ca35091f) Fix "charging keeps stopping at min < capacity < max" (@zhugexinmo)

still doesn't work. steps as above. system shows charging icon, but Acc Settings app shows charging/idle/discharging [acc-logs-joyeuse.zip](https://github.com/VR-25/acc/files/8934351/acc-logs-joyeuse.zip)
I'm lazy to record the video now. Do you have the specific steps / settings so that I can follow?

VR-25, 2022-06-19:
https://github.com/VR-25/acc/releases/tag/v2022.6.19-dev

VR-25, 2022-06-21:
https://github.com/VR-25/acc/releases/tag/v2022.6.21-dev

VR-25, 2022-06-26:
Any update on this?
Yet a new pre-release is up.

vonguyenkhang, 2022-06-26:
> Any update on this? Yet a new pre-release is up.

Sorry I'm very busy recently. I'll give it a try later.

vonguyenkhang, 2022-06-26:
> > Any update on this? Yet a new pre-release is up.
> 
> Sorry I'm very busy recently. I'll give it a try later.
I had a feeling that at first it's ok, but after I unplugged/plugged/stopped/restarted/switched profiles in AccA a lot of times then it's not working again (AccSettings shows Charging/Idle/Discharging while Android shows charging all the time). 
Initial log when everything seems ok: [acc-logs-joyeuse_v2022.6.25-dev_202206250_2220.zip](https://github.com/VR-25/acc/files/8987267/acc-logs-joyeuse_v2022.6.25-dev_202206250_2220.zip)
Log when things become strange: [acc-logs-joyeuse_v2022.6.25-dev_202206250_2220 (2).zip](https://github.com/VR-25/acc/files/8987268/acc-logs-joyeuse_v2022.6.25-dev_202206250_2220.2.zip)

Can I continue using AccA to switch charging profile, or I should stop using AccA? 
Is AccSettings app compatible with these dev releases?
Can I use Android batt status as the "ground truth" to determine if it's charging or not charging/idle?
How to determine if it's in "battery idle" mode? (I used to combine AccA status "Discharging" + AccA current 0mA + Android batt status not charging to conclude that it's in idle mode)

VR-25, 2022-06-28:
https://forum.xda-developers.com/t/advanced-charging-controller-acc.3668427/post-87084881

VR-25, 2022-07-04:
https://github.com/VR-25/acc/releases/download/v2022.7.4-dev/acc_v2022.7.4-dev_202207040_1927.zip

vonguyenkhang, 2022-07-09:
> https://github.com/VR-25/acc/releases/download/v2022.7.4-dev/acc_v2022.7.4-dev_202207040_1927.zip

- Idle mode stopped working ~30s after reaching terminal charing percentage: there's unplugged sound and battery % really drops after a while.
- Small difference with v2022.2: Charging current limit doesn't follow setting as closely as v2022.2, but this may not be the real issue because my charging current is very low (<500mA) and my old settings may not be compatible will new acc.

rafifos, 2022-07-10:
I had the same issue, turns out that AccA was using it's bundled (old) version of acc, just point it to the v2022.7.4-dev version and you're good to go.

vonguyenkhang, 2022-07-10:
> I had the same issue, turns out that AccA was using it's bundled (old) version of acc, just point it to the v2022.7.4-dev version and you're good to go.

You're right! Thanks for posting your finding!

VR-25, 2022-07-11:
Check the releases page. There's a new pre-release.

vonguyenkhang, 2022-07-11:
> Check the releases page. There's a new pre-release.
> […](#)

All good!

closed 2022-07-11 by vonguyenkhang