VR-25/acc #158: No longer limiting charge

mercertom, 2022-06-28:
Lately ACC's been "charging rapidly" and getting super hot, despite being up to date and active. Today updated Magisk, hoping that might fix things, but no such luck.

Used to limit charge current, now it doesn't. Not clear that anything changed, though perhaps this started at an update to ACC?

mercertom, 2022-06-28:
Perhaps, but if so, ACC limited charge for months after upgrade, then abruptly stopped limiting charge. I was surprised to see "charging rapidly" for the first time in about a year (since I installed ACC), and it's scorchingly hot about half the times I charge it now.

VR-25, 2022-06-28:
https://forum.xda-developers.com/t/advanced-charging-controller-acc.3668427/post-87084881

mercertom, 2022-06-28:
I don't know what to make of that.

mercertom, 2022-06-28:
Installed the pre-release, and phone still says "charging rapidly", so I assume it's still not limiting charge voltage. Is there any way it might be limiting charge voltage AND displaying "charging rapidly"?

Over the period it seemed to be working (charged slowly, never overheated), it wouldn't display "charging rapidly".

VR-25, 2022-06-29:
"Charging rapidly" is not necessary accurate, and a system update may change this. The most reliable way of checking whether current os actually being limited is `acc -i curr` (or `acc -w0 curr`).

mercertom, 2022-06-29:
where do you enter that command?

mercertom, 2022-06-29:
Where do I enter those commands?

mercertom, 2022-06-29:
Also, I believe it's charging rapidly / high voltage because it overheats like 50%+ the time, just like before I installed acc.

VR-25, 2022-06-29:
You can use any terminal emulator or adb shell.
Alternatively, install one of the front-end apps. They show current, voltage, temperature and other relevant information.

AccSettings (recommended)
https://github.com/CrazyBoyFeng/AccSettings/releases/

AccA (hasn't been updated in ages)
https://github.com/MatteCarra/AccA/releases/

VR-25, 2022-06-29:
What's your current limit?

mercertom, 2022-06-29:
Assuming it was unlimited. Used the recommended GUI to set current to 1000mA.

Then the charge percentage limit was somehow overridden!

mercertom, 2022-07-01:
I was using the last acc version known to work. It just stopped working. I'm doing a manual % stop by having a voltage limit, so it stops +- 1%. Not sure if it's worse for battery health/life for it to hold steady at that voltage (~75%) vs to charge to 75%, deplete to 70%, charge to 75% etc like it does by default.

VR-25, 2022-07-01:
Ss preciously stated, the current limiting code hasn't changed it a very long time.
I'll perfom a deeper inspection elsewhere.
We'll see if the next build makes a difference.

Aasikki, 2022-07-01:
I just installed acc (+AccA) for the first time on my OnePlus 7 Pro running the latest Pixel experience and set capacity limit to 80%. After reaching 80% AccA claims that I'm discharging but the battery percentage keeps rising.

VR-25, 2022-07-01:
This means the  charging switch in use is unreliable.
Run acc -t on terminal to see which switches work, and pick one with acc -ss.
Refer to acc'd README > troubleshooting > charging switch for more details.

mercertom, 2022-07-01:
May be the root of the problem. In termux, "no command acc found".

VR-25, 2022-07-04:
https://github.com/VR-25/acc/releases/download/v2022.7.4-dev/acc_v2022.7.4-dev_202207040_1927.zip

closed 2022-07-29 by VR-25

Iey4iej3, 2022-08-14:
> The problem I'm having might be related to this issue. At first, using acca, I'd let acc automatically choose the switch, which was `battery/batt_slate_mode 0 1`. Then, I realized the battery heated up quite often and charged extremely slowly to the point that it'd discharge if I used my phone plugged in. Thankfully, I tried out a different switch that worked really well.
> 
> For months, I've been using the switch `battery/hmt_ta_charge 1 0` without any issues. Lately, however, it hasn't been functioning properly. It stops at 60% or goes up to 100% although it's been limited at 80%. Once the battery reaches 100%, I can't charge my phone until I restart it. Testing it in acca, it says "an error has occured". At some point, I managed to get the switch working (without doing or modifying anything, really). It's broken back again, though.
> 
> What's strange is nothing has been tampered with during this time. Acc hasn't been updated, my phone doesn't receive updates anyway, and I've made no modifications to my phone.

Are you using a Samsung device? I seem to suffer from something similar. The workaround is to restart the acc daemon, but this has to be done again and again.

yephny, 2022-08-14:
@Iey4iej3 I am, but it's pretty random. I've been happy since the day after I left that comment.

Iey4iej3, 2022-08-14:
Did you try to restrict the current? It does not seem to be working (I am using v2022.7.30) under `hmt_ta_change`. I can make it work via rebooting, but hours later, it breaks again.

yephny, 2022-08-14:
@Iey4iej3 I do, but I don't think it's working, although the test passes.

Iey4iej3, 2022-08-14:
It seems better to reopen this issue.

mercertom, 2022-10-11:
may have discovered the root of the problem. No command acc found (in termux)

-------- Original Message --------

Iey4iej3, 2022-10-11:
> may have discovered the root of the problem. No command acc found (in termux)

How to run `acc` is device specific. On my Samsung tablet, I have to run `/dev/acc` under root, which is a symbolic link to `/data/adv/vr25/acc/acc.sh`. On my smartphone, it is `/system/bin/acc`, a symbolic link to `/data/adb/modules/acc/acc.sh`.

Anyway, this does not lead to what you described.