VR-25/acc #159: Suddenly Doesn't Work

christantoan, 2022-06-28:
Hello,

I'm having a problem with ACC suddenly doesn't work. The only thing I updated today was Shamiko but rolling back its version still doesn't work. I also have tried rolling back ACC to v2022.2.3 but also didn't work.
Some extra info:
I always use "battery/op_disable_charge 0 1 --" because it always worked for me and I want to use Battery Idle mode even though the test sometimes says it doesn't work. One thing I notice is that, in Battery Info page, op_disable_charge is still set to 0 even though it should be set to 1?
I also often have problem with the test because it often loops reporting the current battery status.
[acc-logs-OnePlus7Pro.zip](https://github.com/VR-25/acc/files/9003132/acc-logs-OnePlus7Pro.zip)

Thank you in advance!

VR-25, 2022-06-28:
new build: https://forum.xda-developers.com/t/advanced-charging-controller-acc.3668427/post-87084881

VR-25, 2022-06-28:
See if the issues remain.

> When testing switches, I notice that running acc -d also stops accd but acc -e doesn't run accd Is that normal?

This is intended behavior.
Those two commands must stop accd to avoid interference (manual vs automatic modes).

christantoan, 2022-06-29:
Thanks for the new build. I'll test it for now

christantoan, 2022-06-29:
Seems to work great now
Thanks!

closed 2022-06-29 by christantoan