VR-25/acc #163: Notes on Sony Xperia devices

hellobbn, 2022-07-27:
Hi,

I am creating this issue to provide some information on the specific mechanism sony's own implementation of "battery idle mode" (Which sony calls it H.S. mode).

__TL; DR__
To enable battery idle mode, issue the following command to the kernel interface:
```bash
# Activate smart charging
echo 1 > /sys/class/power_supply/battery_ext/smart_charging_activiation

# Enable battery idle mode
echo 1 > /sys/class/power_supply/battery_ext/smart_charging_interruption
```
To disable:
```
echo 0 > /sys/class/power_supply/battery_ext/smart_charging_interruption
```

__Detail:__

In stock sony devices, this battery idle mode can be enabled in settings, and when it is enabled, it makes a call to sony's charger management HAL, which will set the property `persist.vendor.battery.smt_chg`, and the init file will write to the kernel interface mentioned above when a property change is noticed:

```
on property:persist.vendor.battery.smt_chg=activate
    write /sys/class/power_supply/battery_ext/smart_charging_activation 1

on property:persist.vendor.battery.smt_chg=charging_suspend
    write /sys/class/power_supply/battery_ext/smart_charging_interruption 1

on property:persist.vendor.battery.smt_chg=charging_resume
    write /sys/class/power_supply/battery_ext/smart_charging_interruption 0
```

Is it possible for that to be  implemented in acc?

VR-25, 2022-07-29:
Sure.
ACC supports multiple charging switches (e.g., acc --set charging_switch="sw1 on off sw2 on off...")
New pre-release coming up soon.

closed 2022-07-29 by VR-25

hellobbn, 2022-07-29:
I am sorry, I made a typo here, it is not `smart_charging_activiation` but `smart_charging_activation`

Sorry for the mistake

hellobbn, 2022-07-30:
After the correction, the test does not seem to work properly:

```
6/24: battery_ext/smart_charging_activation 0 1 battery_ext/smart_charging_interruption 0 1
  switch: off (1, 1)	current: 1934mA (Charging)
  switch: off (1, 1)	current: -66mA (Discharging)
  switch: on (1)	current: -66mA (Discharging)
  Switch doesn't work ❌
  ##########
  Waiting for charging to resume...
  If it doesn't happen after a few seconds, try re-plugging the charger.
  If it's taking too long, unplug the charger, stop the test with CTRL-C, run accd -i, wait a few seconds, and retest.
  In extreme cases, one shall comment out (blacklist) this switch in /data/adb/vr25/acc-data/logs/write.log, reboot (to enable charging), and re-run the test.
  ##########
  switch: - (1)	current: -66mA (Discharging)
  switch: - (1)	current: 0mA (Idle)
  switch: - (1)	current: 0mA (Idle)
  switch: - (1)	current: -1mA (Idle)
  switch: - (1)	current: -0mA (Idle)
  switch: - (1)	current: -0mA (Idle)
  switch: - (1)	current: -0mA (Idle)
  switch: - (1)	current: -0mA (Idle)
  switch: - (1)	current: -1mA (Idle)
  switch: - (1)	current: -1mA (Idle)
  switch: - (1)	current: -66mA (Discharging)
  switch: - (1)	current: 0mA (Idle)
  switch: - (1)	current: 0mA (Idle)
  switch: - (1)	current: 0mA (Idle)
  switch: - (1)	current: 0mA (Idle)
  switch: - (1)	current: -1mA (Idle)
  switch: - (1)	current: -1mA (Idle)
  switch: - (1)	current: -1mA (Idle)
  switch: - (1)	current: -0mA (Idle)
  switch: - (1)	current: -0mA (Idle)
  switch: - (1)	current: -0mA (Idle)
```

hellobbn, 2022-07-30:
The problem is possibly because that the interface `/sys/class/power_supply/battery_ext/smart_charging_activation` cannot be set to 0 once it is 1 (And it is expected behavior).

VR-25, 2022-07-30:
Ok, I see. Will push changes accordingly soon.

hellobbn, 2022-07-30:
Thanks, really appreciate it.