VR-25/acc #164: Question: Stop charging at 81%

likkai, 2022-08-05:
I'm on miui 13 (Android 12), I'm just wondering why does the phone stop charging when it reaches 81% even though the pause_capacity was set to 86%. It's not a serious problem, I'm just wondering why did it happen. 
[image]
[image]

ViniciusRed, 2022-10-19:
update your acc to the latest version to see if it fixed
[v2022.7.30-dev](https://github.com/VR-25/acc/releases/tag/v2022.7.30-dev)

likkai, 2022-11-03:
Late reply but yes, it does fix the problem

closed 2022-11-03 by likkai