VR-25/acc #165: acc does not stop charging on Nexus 5 running LineageOS 14.1

bgx90, 2022-08-07:
I  tried [v2022.6.4](https://github.com/VR-25/acc/releases/tag/v2022.6.4) . No matter what the charging switch setting was set to, the charging state kept switching back and forth between charging and not charging so that the battery continued to charge regardless of pause and resume capacity settings. I then tried [v2022.7.30-dev](https://github.com/VR-25/acc/releases/tag/v2022.7.30-dev) and saw the same behavior. I tried all working charging switches and automatic charging switch selection with batt_status_workaround set to true and false, but the phone always kept switching between charging and not charging every minute or so. Attached are the logs from my test of [v2022.7.30-dev](https://github.com/VR-25/acc/releases/tag/v2022.7.30-dev)

[accd-hammerhead.log](https://github.com/VR-25/acc/files/9277494/accd-hammerhead.log)
[acc-hammerhead.log](https://github.com/VR-25/acc/files/9277495/acc-hammerhead.log)
[acc-i.txt](https://github.com/VR-25/acc/files/9277496/acc-i.txt)
[acc-p.txt](https://github.com/VR-25/acc/files/9277497/acc-p.txt)
[acc-t_output-hammerhead.log](https://github.com/VR-25/acc/files/9277498/acc-t_output-hammerhead.log)
[charging-current-ctrl-files.txt](https://github.com/VR-25/acc/files/9277499/charging-current-ctrl-files.txt)
[charging-switches.txt](https://github.com/VR-25/acc/files/9277500/charging-switches.txt)
[charging-voltage-ctrl-files.txt](https://github.com/VR-25/acc/files/9277501/charging-voltage-ctrl-files.txt)
[config.txt](https://github.com/VR-25/acc/files/9277502/config.txt)
[dumpsys-battery.txt](https://github.com/VR-25/acc/files/9277503/dumpsys-battery.txt)
[init.log](https://github.com/VR-25/acc/files/9277504/init.log)
[install.log](https://github.com/VR-25/acc/files/9277505/install.log)
[magisk.log](https://github.com/VR-25/acc/files/9277506/magisk.log)
[power_supply-hammerhead.log](https://github.com/VR-25/acc/files/9277507/power_supply-hammerhead.log)
[power-supply-logger.sh.log](https://github.com/VR-25/acc/files/9277508/power-supply-logger.sh.log)
[ps-blacklist.log](https://github.com/VR-25/acc/files/9277509/ps-blacklist.log)
[write.log](https://github.com/VR-25/acc/files/9277510/write.log)

Ciantic, 2022-08-19:
I have lineageos 18.1 on Moto G5.

It has completly life of it's own.

```
cedric:/ # acc -f 50

Charging enabled until battery capacity >= 50%

cedric:/ # acc --info

...
Uevent
  CAPACITY=32
  ...
  STATUS=Discharging
  
```

It seems to keep the battery at ~30% if I keep the screen open with [AOA](https://play.google.com/store/apps/details?id=com.newgen.alwayson&hl=fi&gl=US)... It might be that acc isn't working at all.

VR-25, 2023-07-16:
[acc_v2023.7.16_202307160.zip](https://github.com/VR-25/acc/files/12063453/acc_v2023.7.16_202307160.zip)
Try this build.

closed 2023-11-30 by VR-25