VR-25/acc #166: Have to manually "accd -i" once each day

seraphg, 2022-08-09:
Hi. First thank you for making acc. It's so much helpful for me and really love it!

For now I have  a question of a strange behavior if you can help: 
ACC works almost perfectly on my Redmi K50p with MTK 9000, even the Idle mode works. But strangely every morning when I use my phone for a while and then plugin the charger, the status stuck in Discharging and the current keeps showing negative values, then the phone will directly charge to 100% regardless my settings (stop at 90%). 
I have to manually do "accd -i", and then it will be working again: the status will change to Charging with positive current values, and it will stop @ 90% then goes to Idle mode). 
I have 'optimized charging' already disabled in Settings. It's quite weird it happens everyday after overnight, not during the day.

Could you pls help? Thank you

Iey4iej3, 2022-08-14:
It seems that I suffer from similar issues on my Samsung Tab S6 Lite, with the stable release in June 2022.

seraphg, 2022-08-14:
I further investigated this problem. It's basically like this:
- Whenever I unplug the charging cable, the status keeps in "Charging", and the current shows Positive values.
- Then when I plug-in the charging cable, the status is still "Charging", and the current shows small Positive values (like 200mA etc.). Then I realized it's actually NOT charging, and the battery keeps draining. And while in this status, no matter how many times I unplug or plug-in the cable, the status is always "Charging" while it's actually always Discharging...
- In order to solve this, I have to unplug the cable (cannot do it while plugged in the cable), then accd -i. Then the status will change to "Discharging" with a Negative current value.
- Then when I plug-in the cable, the status shows Charging correctly, and the current is correct as well. And it can stop at configured percentage then enters Idle mode.
- But whenever I unplug the cable. I have to do above again.
- I'm using the 0730 build.

Hope this descriptions helps..

IMgoRt, 2022-11-11:
Also seems to be required on my Pixel 7 - happily idles for a random period then starts charging while thinking it's discharging

red0fireus, 2023-02-14:
Having a similar issue on the Pixel 7 Pro.

Using the latest beta and my Pixel 7 Pro it will continue to charge to 81% and then stop or fall to 79% charge and stop charging from there.

Running "accd - i" helps fix this issue temporarily.

blksith0, 2023-05-23:
I have to restart the daemon every day with `acc` then option `4`.
This behavior started several months ago.

VR-25, 2023-07-14:
[acc_v2023.7.13-dev_202307130_1925.zip](https://github.com/VR-25/acc/files/12043016/acc_v2023.7.13-dev_202307130_1925.zip)

Do y'all still face those issues with the latest build?

red0fireus, 2023-07-14:
I have not seen the issue since the latest update