VR-25/acc #170: Unexpected reboots on Samsung SM-T580 (gtaxlwifi)

BVollmerhaus, 2022-09-14:
**OS:** LineageOS 18.1 ([Unofficial](https://forum.xda-developers.com/t/lineageos-18-1-for-sm-t580-gtaxlwifi-and-sm-p580-gtanotexlwifi.4274293/))
**ACC Version:** `v2022.7.30-dev`
**Magisk Version:** `25.2`

### Setup

I just set up a Samsung Galaxy Tab A 10.1 (2016) with a clean installation of Lineage 18.1 and installed the latest ACC version via Magisk. Right afterwards, it randomly rebooted once, but I paid it no mind and configured my desired charging limits (`acc 50 45`). This is the only configuration change I made so far.

### Issue

The charging _does_ start and stop at the desired thresholds (45/50), but the device reboots seemingly randomly (irrespective of whether a threshold is reached or the screen being on, for example). For instance, it just now rebooted while sitting at 50 % battery with the screen off (but also does so mid-use). As far as I can see, this _only_ happens when the device is plugged into a charger, not while on battery.

Running `acc -t` results in a reboot pretty much immediately (both with the charger plugged in or removed, although I'm aware the latter isn't supposed to be useful). I tried explicitly testing the `batt_slate_mode` charge switch (presumably used by default on Samsung devices?) via `acc -t "battery/batt_slate_mode 0 1"`, which also leads to an immediate reboot. Testing the only other charging switch–looking thing I could find in sysfs (`acc -t "sm5703-charger/charging_enabled 1 0"`) does as well.

### Progress

To debug this, I manually set the value of the `battery/batt_slate_mode` charging switch via `echo 1 > /sys/class/power_supply/battery/batt_slate_mode`, which successfully starts (`0`) and stops (`1`) the charging without a reboot. `sm5703-charger/charging_enabled` does not appear to be settable (`cannot create <path>: Permission denied`, even when run as root).

Considering ACC starts and stops the charging correctly, could it be that the charging switch actually works properly, but ACC performs some other operation, such as accessing an additional sysfs file, which triggers a reboot? What might be the best way to debug this?

### Attachments

Please see the attached output generated via `acc -le` right after a random reboot occurred: [acc-logs-gtaxlwifi.tgz](https://github.com/VR-25/acc/files/9568654/acc-logs-gtaxlwifi.tgz)

Bryantdl7, 2022-12-27:
Running the official LOS 19.1 on mine I don't get reboots on gta4xlwifi if that helps any, I am using the battery/hmt_ta_charge 1 0 switch.

Dimoon12, 2023-03-19:
reboots at pixel 4xl with linage.

VR-25, 2023-07-15:
Do you still face reboots even with the latest build (#197)?
If so, while plugged, run `acc -x -t` and share `/sdcard/acc-*.log`.

closed 2023-11-30 by VR-25