VR-25/acc #173: Question: Will it interfere with Adaptive Battery Android feature

uriesk, 2022-11-19:
I was wondering if i should tun Adaptive Battery in the settings on or off.
Apparently it makes Android learn the usage behaviour and adjust performance of the device based on when it expects you to charge it again.
https://9to5google.com/2022/09/15/android-adaptive-battery-explainer/

I imagine that it would not do a great job when acc is running.
What is the general suggestion? Leave it on or turn it off?

red0fireus, 2023-02-14:
I had to disable adaptive charging to fix some of my issues.

I've left adaptive battery enabled on the Pixel 7 Pro.

VR-25, 2023-07-16:
You can leave it on.
To have the best of both worlds, you can use the capacity_mask feature. As the name implies, it makes Android "believe" that the battery has been charging to 100%.

[acc_v2023.7.16_202307160.zip](https://github.com/VR-25/acc/files/12063453/acc_v2023.7.16_202307160.zip)
Try this build.
`acc -s cm=true`

closed 2023-07-16 by VR-25