VR-25/acc #174: Hook the current battery percent/capacity given to the OS

devnoname120, 2022-11-22:
**Nice consequences**:
- Fix the Android status bar to show a percent in the range 0-100% as defined by the custom battery range.
- Fix apps that display graphs about energy/phone usage.
- Fix the breaking of the native OS low-battery warning. Stock Android is configured to show it at 20%.
- Fix the breaking of the native OS power saving modes when the battery reaches x%: kernel scheduler settings, background apps limiter, screen brightness/color range tweaks, radio tweaks, etc.
- Let Android gracefully shutdown the phone instead of doing a hard power cut off when reaching e.g. 2%. This avoids corrupting the filesystem and this lets apps know that they should pause what they are doing/save their work.
- Probably many more fixed that I haven't thought about.

brightplastik, 2022-12-10:
Yes, I have very limited understanding of how and why, but in a Mi9 Lite and a Tab S4 they both have scrambled percentage after applying the settings with the panel by CrazyBoyFeng. I wanted 91% max charge, and it shows 78% on the Xiaomi, and 85% on the tablet.

VR-25, 2023-07-14:
[acc_v2023.7.13-dev_202307130_1925.zip](https://github.com/VR-25/acc/files/12043016/acc_v2023.7.13-dev_202307130_1925.zip)

`acc -s cm=true`

closed 2023-07-16 by VR-25