VR-25/acc #178: Charging limit unreliable on Pixel 7 Pro

matejdro, 2022-12-28:
I've been trying to use ACC for a while and charging limit percentage seems very unreliable. Regardless of what switch I use (either one of those that work from the `acc --test` command or native Pixel switch (#124)), charging stop will work at first, but then stop working a while later. It seems like acc disables itself after some time.

Here are logs after an overnight charge that charged to over my limit of 70%: [Archive.zip](https://github.com/VR-25/acc/files/10312559/Archive.zip)

kleops9, 2023-01-05:
Did you try to use the latest [v2022.7.30-dev](https://github.com/VR-25/acc/releases/tag/v2022.7.30-dev) Pre-release module? 
I have pixel 6 pro and had the same issue but with the prelease it works.

matejdro, 2023-01-05:
No, that was on https://github.com/VR-25/acc/releases/tag/v2022.6.4. Will try the prerelease and report, thanks.

matejdro, 2023-01-25:
Pre-release has been working for the past couple of weeks. Thanks!

closed 2023-01-25 by matejdro