VR-25/acc #179: MI 11 Ultra wireless charging not stopping

knasiotis, 2023-01-11:
When I connect a mi 11 ultra to a xiaomi charger with a xiaomi wireless stand, the phone never stops charging. What's interesting is that when I place the phone in the wireless stand, the AccA app shows the phone as "discharging" -6000mah but I can see the battery percentage increase over time. Is there a way to export the logs from the phone?

knasiotis, 2023-01-11:
I uploaded the logs to your dropbox

knasiotis, 2023-04-11:
Still shows a negative current btw

cubale, 2023-05-28:
I have the same problem.

VR-25, 2023-07-15:
Update [acc]([acc_v2023.7.13-dev_202307130_1925.zip](https://github.com/VR-25/acc/files/12043016/acc_v2023.7.13-dev_202307130_1925.zip)).

knasiotis, 2023-07-16:
@VR-25 This build works great so far, I tried to save logs but I couldn't reproduce the issue. 
The behavior was (when battery limit was reached):
-Status: discharging
-Wireless stand was active
-Phone was charging above limit
Now it is:
-Status: discharging
-Wireless stand tries to make nfc handshake
-It succeeds for a split second and then stops
-Phone is not charging

So it seems to be working now, I haven't seen the battery going above the limit. I don't have the time to test it extensively but I will inform you when I do it

closed 2023-07-29 by VR-25