VR-25/acc #182: Charge current limit does not work on Samsung S6 Lite (gta4xlwifi)

Iey4iej3, 2023-01-25:
2022-07-30 build of acc.

Attach the logs: 
[acc-logs-gta4xlwifi.tar.gz](https://github.com/VR-25/acc/files/10500093/acc-logs-gta4xlwifi.tar.gz)

ivorget, 2023-01-27:
FWIW, on my s4 mini, the charging strength is a percentage setting controlled by siop_level. And I didn't find any reference to siop_level in the ACC code.

(I haven't used ACC recently, and I think it crashed my phone when I tried it years ago.)

VR-25, 2023-07-14:
> 2022-07-30 build of acc under `gta4xlwifi`.
> 
> Attach the logs: [acc-logs-gta4xlwifi.tar.gz](https://github.com/VR-25/acc/files/10500093/acc-logs-gta4xlwifi.tar.gz)

Current and voltage control are not supported by all devices/kernels.

VR-25, 2023-07-14:
> FWIW, on my s4 mini, the charging strength is a percentage setting controlled by siop_level. And I didn't find any reference to siop_level in the ACC code.
> 
> (I haven't used ACC recently, and I think it crashed my phone when I tried it years ago.)
 
Do you happen to remember the path to the siop_level control file? I can add support for it.

closed 2023-07-14 by VR-25

Iey4iej3, 2023-07-14:
> Do you happen to remember the path to the `siop_level` control file? I can add support for it.

I performed `find`, and found it on my Samsung S6 Lite: `/sys/devices/platform/battery/power_supply/battery/siop_level`. After googling, it is a percentage, whose default level is 100.

VR-25, 2023-07-14:
I'll add it as part of the `temp_level` feature, which also ranges from 0 to 100%.

VR-25, 2023-07-16:
[acc_v2023.7.16_202307160.zip](https://github.com/VR-25/acc/files/12063453/acc_v2023.7.16_202307160.zip)
`acc -s tl=50 # ranges from 0-100`

Iey4iej3, 2023-09-09:
> I'll add it as part of the `temp_level` feature, which also ranges from 0 to 100%.

This seems to be contradictory. In the readme, it's said that, when `temp_level` is set to be 100, it blocks the current. However, `siop_level` is 100% when there is no restriction.

VR-25, 2023-09-09:
`temp_level` always ranges from 0-100 and it automatically flips the values to be compatible with siop_level (i.e., if you set `temp_level=100`, `siop_level` will be set to `0`).

VR-25, 2023-09-09:
Install the latest snapshot.
[acc_v2023.9.9-dev_202309090_2003.zip](https://github.com/VR-25/acc/files/12566528/acc_v2023.9.9-dev_202309090_2003.zip)