VR-25/acc #183: force single charge will not wait at target capacity when unplugged

matejdro, 2023-01-30:
It seems like charge to full capacity once (`acc -f 100`) does not wait for the charger to get unplugged before it re-triggers charging limits. This results in a following:

1. Let's say I plug the phone overnight before a trip, so I want 100% charge
2. I set acc to force charge to 100% once and plug it in
3. In the middle of the night, phone will reach 100% capacity
4. Force charge will disengage
5. Since phone's capacity (100%) is over currently set pause level, battery will start draining
6. Depending on what tasks phone was performing overnight, I might end up with a phone well below target 100% battery in the morning.

red0fireus, 2023-02-11:
I'm wondering how to do this too.

VR-25, 2023-07-15:
> ...does not wait for the charger to get unplugged...

This is a limitation most devices have, unfortunately.
ACC can't detect whether the charger is truly unplugged, simply because most devices don't provide that information.

The best you can do is add a delay before the command, e.g., `sleep $(3600 * 6) && acc -f # Wait for 6 hours before running acc -f`.

closed 2023-07-15 by VR-25

matejdro, 2023-07-16:
But, from what I understand, when force charge is enabled, you CAN detect when it is charging or not? Since there are no limits applied, phone behaves normally  and thus regular Android APIs for charging/discharging should work.

VR-25, 2023-07-16:
If you unplug before force charge disengages, it will keep waiting until battery level reaches the desired value.

Once it has disengaged, the usual settings are applied, regardless of whether the charger is still plugged. Things would have been different if Android had a standard "plug state" API.

I'll implement the desired change. If your device supports it, it'll behave exactly as you want. Will share the build here soon.

VR-25, 2023-07-16:
[acc_v2023.7.16_202307160.zip](https://github.com/VR-25/acc/files/12063453/acc_v2023.7.16_202307160.zip)