VR-25/acc #185: Capacity sync account for shutdown capacity

TheHardew, 2023-02-23:
Please add an option to account for shutdown capacity in capacity sync:

`capacity = 100(capacity - shutdown_capacity) / (pause_capacity - shutdown_capacity)`

Recently my phone got stuck in a boot loop after discharging. To make sure that does not happen any more, I'm using shutdown_capacity to prevent the phone from discharging completely. But that means, that some percentage of the displayed battery is not available and I'd like capacity sync to account from that.

I have not actually tested whether that's how it's already implemented, but from `README.md` and the source code I believe it has not been implemented.

VR-25, 2023-06-25:
What if I made capacity_mask account for the difference between `pause_capacity` and `shutdown_capacity` instead?

```
# capacity_mask (cm) # Default: false
# Implies capacity_sync.
# This forces Android to report "capacity = capacity * 100 / pause_capacity", effectively masking capacity limits.
# It also prevents Android from getting capacity readings below 2%, since some systems shutdown before battery level actually drops to 0%.
# Use case: secretly install acc on a relative's device, and enable this, so that they see pause_capacity as 100%.
```
With the proposed change, we would have `capacity = capacity * 100 / (pause_capacity - shutdown_caapcity)`.

VR-25, 2023-07-14:
[acc_v2023.7.13-dev_202307130_1925.zip](https://github.com/VR-25/acc/files/12043016/acc_v2023.7.13-dev_202307130_1925.zip)

`acc -s cm=true`

closed 2023-07-14 by VR-25