VR-25/acc #187: How to set maxChargingCurrent CTRL_FILE correctly?

nabakdev, 2023-03-03:
i have build manually through latest github source (commit [978740f](https://github.com/VR-25/acc/commit/978740f871cc72d4f7cd9b0bf25cce38da788853))

then modify this file `install/default-config.txt`

```bash
capacity=(10 50 75 79 auto false)
chargingSwitch=(/sys/class/power_supply/battery/battery_charging_enabled 1 0)
maxChargingCurrent=(500 /sys/class/power_supply/battery/constant_charge_current_max::500000::-22)
```

the problem is i can't limit max current, `acc` will rewrite the `maxChargingCurrent` value with another CTRL_FILE(which is that CTRL_FILE doesn't work) when i run `acc -s mcc=500`

how to set `maxChargingCurrent` CTRL_FILE permanently even after reboot?

DEVICE: XIAOMI POCO F4 5G (munch)

NOTE:
- `/sys/class/power_supply/battery/constant_charge_current_max` is the only one that works to limit current 

for now i need to change this file `/sys/class/power_supply/battery/constant_charge_current_max` value manually to limit charging current

*sorry for my bad english

nabakdev, 2023-06-19:
i've found that my ctrl file has negative value as default so acc blacklist it.

closed 2023-06-19 by nabakdev

sourcevault, 2024-11-16:
> * /sys/class/power_supply/battery/constant_charge_current_max

Hi, how were you able to manually change this value ? I get this error on my redmi 12 4G.
```
tapas:/ # 500 >> /sys/class/power_supply/battery/constant_charge_current_max
/system/bin/sh: 500: inaccessible or not found
```

nabakdev, 2024-12-06:
> > * /sys/class/power_supply/battery/constant_charge_current_max
> 
> Hi, how were you able to manually change this value ? I get this error on my redmi 12 4G.
> 
> ```
> tapas:/ # 500 >> /sys/class/power_supply/battery/constant_charge_current_max
> /system/bin/sh: 500: inaccessible or not found
> ```

here's the correct command
```
echo 500 > /sys/class/power_supply/battery/constant_charge_current_max
```

sourcevault, 2024-12-27:
> here's the correct command
>
> ```
> echo 500 > /sys/class/power_supply/battery/constant_charge_current_max
> ```

It works for tapas ( redmi note 12 4G ) but not for even ( realme C25s ) it says "Permission Denied".

[image]