VR-25/acc #197: Bootloop when installing on KernelSU

ElDavoo, 2023-07-01:
Hello,
I have an apollopro with crdroid 7.6 and BaalamKernel 5.0 and KernelSU 11041. With magisk the module was working; however, installing acc with KernelSU active makes the device bootloop. I'm testing commit [12864f4](https://github.com/VR-25/acc/commit/12864f4070b0f962d3e5ce9fdb94e2b5f87a29a6). The problem is, I can't debug a lot as I can't open a shell or execute adb operations in bootloop:
```
adb shell
error: Exec '/system/bin/sh' failed: Permission denied
```
I have to use ksu's safe mode to uninstall acc and let everything work again.

VR-25, 2023-07-01:
Do you have `/data/adb/vr25/acc-data/logs/init.log`?

ElDavoo, 2023-07-01:
[init.log](https://github.com/VR-25/acc/files/11927146/init.log)

ElDavoo, 2023-07-01:
I can get access by running `adb root`.
I am not sure if this is a acc problem or a kernelsu problem.
[logcat.txt](https://github.com/VR-25/acc/files/11927199/logcat.txt)
[dmesg.txt](https://github.com/VR-25/acc/files/11927200/dmesg.txt)

VR-25, 2023-07-01:
Ok, can you grab `/dev/.vr25/acc/accd-*.log`.with rooted adb?

ElDavoo, 2023-07-01:
> Ok, can you grab `/dev/.vr25/acc/accd-*.log`?

there is nothing there

ElDavoo, 2023-07-01:
this is after i install acc from the shell script without rebooting
[acc-apollo.txt](https://github.com/VR-25/acc/files/11927294/acc-apollo.txt)

VR-25, 2023-07-02:
[acc_v2023.7.2-dev_202307020_0723.zip](https://github.com/VR-25/acc/files/11928815/acc_v2023.7.2-dev_202307020_0723.zip)
Try that.

ElDavoo, 2023-07-02:
Module installs, but disappears after reboot (not listed in manager and no /dev/.vr25)
[init.log](https://github.com/VR-25/acc/files/11929101/init.log)
[install.log](https://github.com/VR-25/acc/files/11929102/install.log)

VR-25, 2023-07-02:
Weird...

But does it work before rebooting? Since executables are unavailable before rebooting, you have to use the `/dev/` prefix, e.g., `/dev/accd,`, `/dev/accd.`, `/dev/acc -i`.
What do you have in `/data/adb/modules/acc/` right after installing? Is there a `/data/adb/modules_update/acc/` folder as well?

ElDavoo, 2023-07-02:
Yep it seems to work, i'm trying the test charging switches option and seems like it's doing its stuff.
the `/data/adb/modules/acc/` is full of stuff, while the modules_update folder is empty.
KernelSU works with capabilities and namespaces, and I guess this makes everything very confusing. I can also see the module from the manager's list if i force close the app (without rebooting).

VR-25, 2023-07-02:
Is there an `update` file in `/data/adb/modules/acc/`?

ElDavoo, 2023-07-02:
no

VR-25, 2023-07-03:
[acc_v2023.7.3-dev_202307030_2053.zip](https://github.com/VR-25/acc/files/11940609/acc_v2023.7.3-dev_202307030_2053.zip)

ElDavoo, 2023-07-03:
```
_ __       
/\ /\___ _ __ _ __   ___| / _\/\ /\ 
/ //_/ _ \ '__| '_ \ / _ \ \ \/ / \ \
/ __ \  __/ |  | | | |  __/ |\ \ \_/ /
\/  \/\___|_|  |_| |_|\___|_\__/\___/

- Preparing image
- Module size: 430.11 kB
- Estimated minimum size of the filesystem: 11614
- Target image size: 280.76 MB
resize2fs 1.46.2 (28-Feb-2021)
- Mounting image
- Device is system-as-root

Advanced Charging Controller (ACC) v2023.7.3-dev (202307030)
Copyright 2017-2023, VR25
GPLv3+

Installing in /data/adb/modules/acc/...

Error: Failed to install module script
- Error: Failed to install module script
```

VR-25, 2023-07-04:
Do other Magisk modules also disappear on reboot?

ElDavoo, 2023-07-04:
None that i know of (i only have zygisksu and lsposed and they seem to work normally).

SK00RUPA, 2023-07-04:
In addition to the above - Shamiko, Universal SafetyNet Fix, YouTube RVX, MiXplorer also stay after a reboot.
Attaching dmesg, ksu_tree and logcat grabbed right after trying to install the latest failing version `202307030_2053`.
Let me know if you need logs with other versions, like `202307020_0723`.
[](url)
[accbug.zip](https://github.com/VR-25/acc/files/11946379/accbug.zip)

VR-25, 2023-07-04:
Do you have `/data/adb/vr25/acc-data/logs/install.log` from the failed installation?

ElDavoo, 2023-07-04:
[install.log](https://github.com/VR-25/acc/files/11951233/install.log)

SK00RUPA, 2023-07-04:
[install.log](https://github.com/VR-25/acc/files/11951385/install.log)

VR-25, 2023-07-04:
[acc_v2023.7.4-dev_202307040_1852.zip](https://github.com/VR-25/acc/files/11951554/acc_v2023.7.4-dev_202307040_1852.zip)

SK00RUPA, 2023-07-04:
Installed successfully, but unfortunately still disappears after a reboot.
[install.log](https://github.com/VR-25/acc/files/11951715/install.log)
[init.log](https://github.com/VR-25/acc/files/11951716/init.log)

VR-25, 2023-07-04:
[acc_v2023.7.4-dev_202307040_2041.zip](https://github.com/VR-25/acc/files/11952014/acc_v2023.7.4-dev_202307040_2041.zip)

SK00RUPA, 2023-07-04:
Hate to say that, but error during the install once again.
[KernelSU_install_log_2023-07-04-21-58-03.log](https://github.com/VR-25/acc/files/11952071/KernelSU_install_log_2023-07-04-21-58-03.log)
[install.log](https://github.com/VR-25/acc/files/11952072/install.log)

VR-25, 2023-07-04:
[acc_v2023.7.4-dev_202307040_2105.zip](https://github.com/VR-25/acc/files/11952098/acc_v2023.7.4-dev_202307040_2105.zip)

SK00RUPA, 2023-07-04:
Installed successfully and the module disappeared after a reboot in the KernelSU list. However, `/data/adb/vr25` survived and daemon is accessible from e.g. ACC Settings app. `install-tarball.sh-2.log` grabbed after the second reboot.
[install.log](https://github.com/VR-25/acc/files/11952192/install.log)
[install-tarball.sh.log](https://github.com/VR-25/acc/files/11952193/install-tarball.sh.log)
[install-tarball.sh-2.log](https://github.com/VR-25/acc/files/11952194/install-tarball.sh-2.log)
[init.log](https://github.com/VR-25/acc/files/11952191/init.log)

VR-25, 2023-07-04:
[acc_v2023.7.4-dev_202307040_2218.zip](https://github.com/VR-25/acc/files/11952346/acc_v2023.7.4-dev_202307040_2218.zip)

SK00RUPA, 2023-07-04:
Installed and disappeared after a reboot.
Daemon is not accessible & ACC Settings reports `Exit code: 127`. 
`/data/adb/vr25`survived.
 `/dev/.vr25` contains populated `busybox` directory, and almost empty `acc` directory (`plugins`, `acc.lock` and device specific log).

[init.log](https://github.com/VR-25/acc/files/11952511/init.log)
[install-tarball.sh.log](https://github.com/VR-25/acc/files/11952512/install-tarball.sh.log)
[install.log](https://github.com/VR-25/acc/files/11952513/install.log)
[install-tarball.sh-2.log](https://github.com/VR-25/acc/files/11952516/install-tarball.sh-2.log)

VR-25, 2023-07-05:
[acc_v2023.7.5-dev_202307050_2059.zip](https://github.com/VR-25/acc/files/11961494/acc_v2023.7.5-dev_202307050_2059.zip)

If the above fails, try 
[acc_v2023.7.5-dev_202307050_2039.zip](https://github.com/VR-25/acc/files/11961499/acc_v2023.7.5-dev_202307050_2039.zip)

ElDavoo, 2023-07-05:
> [acc_v2023.7.5-dev_202307050_2059.zip](https://github.com/VR-25/acc/files/11961494/acc_v2023.7.5-dev_202307050_2059.zip)
- Bootloop
- [1 - logcat.txt](https://github.com/VR-25/acc/files/11961615/1.-.logcat.txt)
- Before reboot, everything works
- After reboot:
  - /dev/.vr25/acc/ is empty
  - /data/adb/modules/acc/ is full of files (and update file is present - i tried removing it just in case but nothing happened)
  - [1 - init.log](https://github.com/VR-25/acc/files/11961676/1.-.init.log)
  - [1 - install.log](https://github.com/VR-25/acc/files/11961677/1.-.install.log)

> If the above fails, try [acc_v2023.7.5-dev_202307050_2039.zip](https://github.com/VR-25/acc/files/11961499/acc_v2023.7.5-dev_202307050_2039.zip)
- Before reboot, everything works
- After reboot, reboots again ([2 - logcat.txt](https://github.com/VR-25/acc/files/11961742/2.-.logcat.txt))
```
apollo:/data/adb/modules/acc # ls -l
total 9
-rw------- 1 root root  494 2023-07-05 22:27 module.prop
-rwx------ 1 root root  527 2023-07-05 22:27 service.sh
drwx------ 3 root root 1024 2023-07-05 22:27 system
-rwx------ 1 root root 2039 2023-07-05 22:27 uninstall.sh
```
The rest of the files are in /data/adb/vr25/acc/
- After that, it boots again, but I guess KernelSU's primary safety mechanism activated, because the module in the manager seems enabled (but it's actually not - no /dev/.vr25 , no acc command ...). The switch is not toggleable and i can only uninstall the module, so it seems like here KernelSU detects the bootloop and disables the module (which is what is advertised to do in its wiki). Rebooting again does not change anything.
 ```
apollo:/data/adb/modules/acc # ls -l
total 3
-rw-r--r-- 1 root root 494 2023-07-05 22:27 module.prop
-rw-r--r-- 1 root root   0 2023-07-05 22:27 update
```
[2 - init.log](https://github.com/VR-25/acc/files/11961862/2.-.init.log)
[2 - install.log](https://github.com/VR-25/acc/files/11961863/2.-.install.log)

SK00RUPA, 2023-07-05:
For me it's the same - both bootloops.
However we have something, the module itself isn't disappearing after booting in safe mode.

VR-25, 2023-07-05:
[acc_v2023.7.5-dev_202307050_2156.zip](https://github.com/VR-25/acc/files/11962021/acc_v2023.7.5-dev_202307050_2156.zip)

ElDavoo, 2023-07-05:
bootloop

ElDavoo, 2023-07-05:
[logcat.txt](https://github.com/VR-25/acc/files/11962090/logcat.txt)
[init.log](https://github.com/VR-25/acc/files/11962091/init.log)
[install.log](https://github.com/VR-25/acc/files/11962093/install.log)

VR-25, 2023-07-06:
[acc_v2023.7.6-dev_202307060_2024.zip](https://github.com/VR-25/acc/files/11971569/acc_v2023.7.6-dev_202307060_2024.zip)

I added a 3 minutes accd boot delay and set the daemon log file path to `/data/adb/accd-*.log`, so that it persists across reboots.

ElDavoo, 2023-07-06:
bootloop
content of /dev/.vr25/
```
total 0
drwxrwxrwx 2 root root   60 1972-10-25 00:23 acc
drwx------ 2 root root 7220 1972-10-25 00:23 busybox

/dev/.vr25/acc:
total 4
-rw-rw-rw- 1 root root 5 1972-10-25 00:23 acc.lock

/dev/.vr25/busybox:
total 0
lrwxrwxrwx 1 root root 25 1972-10-25 00:23 [ -> /data/adb/ksu/bin/busybox
....
```
[init.log](https://github.com/VR-25/acc/files/11971696/init.log)
[install.log](https://github.com/VR-25/acc/files/11971697/install.log)
[accd-apollo.log](https://github.com/VR-25/acc/files/11971698/accd-apollo.log)

VR-25, 2023-07-06:
[acc_v2023.7.6-dev_202307060_2053.zip](https://github.com/VR-25/acc/files/11971790/acc_v2023.7.6-dev_202307060_2053.zip)

ElDavoo, 2023-07-07:
bootloop
[init.log](https://github.com/VR-25/acc/files/11977371/init.log)
[install.log](https://github.com/VR-25/acc/files/11977372/install.log)
[logcat.txt](https://github.com/VR-25/acc/files/11977373/logcat.txt)

VR-25, 2023-07-08:
Could you run `mounts > /sdcard/mounts.txt` and share the file?

GuruRinpocheFan, 2023-07-09:
Hello,

I had the same problem: bootlooping or module disappearing.
I tried a lot and found a solution: if I first install under KernelSU the module "BusyBox für Android NDK" from Osm0sis, than it works! 😊

As soon as I delete or disable the module, I get bootloops again. So I just leave it on and installed. 

Hope I was able to help (this is my first Github comment ever) 

BTW: With surya (Poco X3 NFC) the newest version 2023-06-26 isn't working stable, sometimes charging doesn't stop, sometimes it doesn't restart, so I went back to the version that comes with AccA (2021-08-31) and so far it works : -) 

BTW2: thanks a lot for the acc module and AccA to everyone involved! 😃

VR-25, 2023-07-09:
Have you tried the latest KernelSu release?

Mylinde, 2023-07-10:
> Could you run `mount > /sdcard/mounts.txt` and share the file?

Same issue here. Crdroid 9.6 2023.06.30, Kernel HYDRA-KSU-OXYFR3, latest daemon from above. My Mount attached.
[mount.txt](https://github.com/VR-25/acc/files/12002380/mount.txt)

ElDavoo, 2023-07-11:
why was the comment about busybox by osm0sis deleted?

VR-25, 2023-07-11:
> why was the comment about busybox by osm0sis deleted?

No idea...
Looks like the person who comented was also the one who deleted it.

ElDavoo, 2023-07-12:
They say their comments keep disappearing, so I will write their comment below.

> if I first install under KernelSU the module "BusyBox for Android NDK" from Osm0sis, than it works! 
> 
> As soon as I delete or disable the module, I get bootloops again. So I just leave it on and installed.
> 
> BTW: With surya (Poco X3 NFC) the newest version 2023-06-26 isn't working stable, sometimes charging doesn't stop, sometimes it doesn't restart, so I went back to the version that comes with AccA (2021-08-31) and so far it works : -)
> 
> BTW2: thanks a lot for the acc module and AccA to everyone involved!
> 
> BTW4: if you are a non-gki KernelSU user where KSU Safe mode isn't implemented, I found out if you have a bootloop you can eighter delete /data/adb/ksu/modules.img or before you try acc (when it is still booting up) make a copy of a working modules.img file and when you have a bootloop, go to your recovery (like twrp) and copy it back to /data/adb/ksu/modules.img
> That way you save a lot of time new installing all modules

I just delete `/data/adb/vr25` and `/data/adb/modules/acc` and it works again.

ElDavoo, 2023-07-12:
I can confirm that with the busybox module things get better:
- Non working builds will not bootloop the device but just disappear after reboot
I checked every build before acc_v2023.7.5-dev_202307050_2156.zip and they don't work, but this one does. The last build (acc_v2023.7.6-dev_202307060_2053.zip) works as well.

Mylinde, 2023-07-12:
> I can confirm that with the busybox module things get better:
>
> * Non working builds will not bootloop the device but just disappear after reboot
> * I checked every build before [...]

Which busybox module do you use? I'm using this https://github.com/Magisk-Modules-Alt-Repo/BuiltIn-BusyBox and have bootoop in normal mode with active acc. Booting with deactivated module and enabling after boot completed works the latest daemon well.

ElDavoo, 2023-07-12:
> Which busybox module do you use?

Busybox for Android NDK 1.34.1 by osm0sis

VR-25, 2023-07-12:
Could one of you share the output of `ls -lR /data/adb/ksu`?

ElDavoo, 2023-07-12:
> Could one of you share the output of `ls -lR /data/adb/ksu`?
With busybox module and acc
```
total 277642
drwx------ 2 root root      3488 2023-07-06 13:31 bin
drwx------ 2 root root      3488 1970-01-01 01:07 log
-rw------- 1 root root 284016640 2023-07-12 17:25 modules.img

/data/adb/ksu/bin:
total 2304
-rwxr-xr-x 1 root root 2107848 2023-07-06 13:31 busybox
lrwxrwxrwx 1 root root      18 2023-07-06 13:31 ksud -> /data/adb/ksud
-rwxr-xr-x 1 root root  247168 2023-07-06 13:31 resetprop

/data/adb/ksu/log:
total 8548
-rw------- 1 root root 4444029 2023-07-12 17:25 boot.log
-rw------- 1 root root 4299484 2023-07-12 13:22 boot.old.log
```

ElDavoo, 2023-07-12:
```
                           _ __       
  /\ /\___ _ __ _ __   ___| / _\/\ /\ 
 / //_/ _ \ '__| '_ \ / _ \ \ \/ / \ \
/ __ \  __/ |  | | | |  __/ |\ \ \_/ /
\/  \/\___|_|  |_| |_|\___|_\__/\___/

- Preparing image
- Module size: 430.35 kB
- Estimated minimum size of the filesystem: 11616
- Target image size: 280.76 MB
resize2fs 1.46.2 (28-Feb-2021)
- Mounting image
- Device is system-as-root

Install busybox or simply place it in /data/adb/vr25/bin/

Error: Failed to install module script
- Error: Failed to install module script

```

VR-25, 2023-07-12:
Got install.log?

ElDavoo, 2023-07-13:
bootloop

Mylinde, 2023-07-13:
Can you take a look to /data/adb/modules/acc/write-config.sh it flails. Maybe that cause the bootloop if it also happens while booting.

[image]

VR-25, 2023-07-13:
> Can you take a look to /data/adb/modules/acc/write-config.sh it flails. Maybe that cause the bootloop if it also happens while booting.
> 
> [image]

It's failing, because it's running out of context. That's not meant to run by itself. It's not a standalone script.

Mylinde, 2023-07-13:
Maybe solved. Check the permissions of the directory's in /data/adb/modules/acc/ they are all rwx - - we need rwx r-x r-x. Changed them and 2023.7.6 works without bootloop.

VR-25, 2023-07-13:
> Maybe solved. Check the permissions of the directory's in /data/adb/modules/acc/ they are all rwx - - we need rwx r-x r-x. Changed them and 2023.7.6 works without bootloop.
> 
> ... still works after some restarts. 👍

Nice!
But one question remains, what's the deal with Osm0sis' BusyBox?

Mylinde, 2023-07-13:
> > Maybe solved. Check the permissions of the directory's in /data/adb/modules/acc/ they are all rwx - - we need rwx r-x r-x. Changed them and 2023.7.6 works without bootloop.
> > ... still works after some restarts. +1
> 
> Nice! But one question remains, what's the deal with Osm0sis' BusyBox?

I don't know.

ElDavoo, 2023-07-13:
> Maybe solved. Check the permissions of the directory's in /data/adb/modules/acc/ they are all rwx - - we need rwx r-x r-x. Changed them and 2023.7.6 works without bootloop.

As a side info, I set the busybox binary permission to rwx r-x r-x , but after a reboot and then reboot to safe mode, they were set to rwx --- --- .

ElDavoo, 2023-07-13:
> Unfortunately, I don't currently have a phone supported by KernelSu.

We can organize an anydesk session maybe?

[7 more replies]

closed 2023-07-14 by VR-25