VR-25/acc #199: Updated capacity_mask logic shows full charge prematurely

kaljade, 2023-07-12:
The latest update improves the capacity_mask logic by showing discharge all the way down to zero, but unfortunately there's a slight error in the new logic that reports the battery at 100% before it has reached the pause_capacity

`# This forces Android to report "capacity = capacity * 100 / (pause_capacity - shutdown_capacity)", effectively masking capacity limits.`

I believe the correct logic should be:

`# This forces Android to report "capacity = (capacity - shutdown _capacity) * 100 / (pause_capacity - shutdown_capacity)", effectively masking capacity limits.`

Currently if you have pause_capacity set at 70 and shutdown_capacity set at 5 then capacity_mask will display 100% at 65 not 70

VR-25, 2023-07-13:
[acc_v2023.7.13-dev_202307130_1925.zip](https://github.com/VR-25/acc/files/12043016/acc_v2023.7.13-dev_202307130_1925.zip)

closed 2023-07-13 by VR-25

kaljade, 2023-07-13:
Installed and confirmed working as intended, thank you for the fast turnaround on the fix and for this incredibly useful module. Cheers, Kal