VR-25/acc #202: ACC randomley stops working

eugenesan, 2023-07-21:
Since trying v2023.7.17_202307170 ACC randomly quits working.

Reverting to previously working versions and configs didn't help.
The only other change that happened was update to Google July update.
Also, the same independently happened on two identical devices.

Here is the end of accd-device.log:
```
132: not_charging
18: typeset 'i='
19: typeset 'nci=15'
20: typeset 'switch=off'
20: flip=''
0: cat /dev/.vr25/acc/.curr
21: typeset 'curThen=506592'
22: typeset 'idleThreshold=40'
23: typeset 'battStatusOverride='
24: typeset 'battStatusWorkaround=true'
26: tt sm7150_bms/charge_disable '*\ --'
386: eval $'case "sm7150_bms/charge_disable" in\n    *\\ --) return 0;;\n  esac'
389: return 1
26: battStatusOverride=''
28: '[' battery/current_now '=' /dev/.vr25/acc/.dummy-curr ']'
29: '[' 1000000 -eq 1000 ']'
29: idleThreshold=40000
32: '[' -z '' ']'
32: '[' -n off ']'
0: seq 15
34: '[' off '=' off ']'
35: status
84: typeset 'i='
0: cat battery/current_now
85: typeset 'curNow=0'
87: read_status
0: cat sm7150_bms/status
55: typeset 'status=Not charging'
58: printf Idle
87: _status=Idle
89: '[' -n '' ']'
95: true
96: '[' Idle '=' Idle ']'
99: '[' -z '' ']'
102: '[' Discharging '!=' Idle ']'
102: '[' DischargingDischarging '!=' Idle ']'
102: '[' Idle '!=' Idle ']'
102: return 0
35: return 0
147: false
164: set +eux
165: eval '${accf-}'
165:
168: '[' -n '' ']'
203: false
203: print_charging_disabled
/data/adb/vr25/acc/accd.sh[643]: print_charging_disabled: inaccessible or not found
203: exxit
6: exit 127
```

VR-25, 2023-07-21:
See if running the following solves the problem: `sed -i 's/^isAccd=false$/: ${isAccd:=false}/' /data/adb/vr25/acc/misc-functions.sh; accd`

eugenesan, 2023-07-21:
> See if running the following solves the problem: `sed -i 's/^isAccd=false$/: ${isAccd:=false}/' /data/adb/vr25/acc/misc-functions.sh; accd`

I've applied the fix. I'll report back to see if it helped.
Due to the randomness of the issue it might take a day.

eugenesan, 2023-07-23:
@VR-25 
The fix seems to be working.
Thanks!

closed 2023-07-26 by VR-25