VR-25/acc #203: Issues with charging thresholds and ACC

tamas646, 2023-07-26:
I've been using ACC for a long time. When I first installed it I was using LOS 19.1 (Android 12), later I upgraded to LOS 20 (Android 13), and everything was working fine (with only a few exceptions). But when I upgraded to the latest ACC build (2023.7.17) things started to break...

I have two devices:
- Moto G6 Plus (evert), LOS 20
- Galaxy Tab S6 Lite (gta4xlwifi), LOS 20

Both devices run the same LOS and ACC versions (latest of both).

I first noticed that charging thresholds are stopped working on both devices. I usually charge my devices at night up to 85% but now they charge up to 100% which is annoying. On my phone (evert) I have an app that logs battery events. From that log I could deduce that the system re-enables charging a few seconds after ACC disables it. So I ran `acc -t` to test the available charging switches to see what's working and switch to a working one. Running `acc -t` I got message 'Terminated'. I tried to reboot the device, tried `accd -i`, and sometimes it was working but eventually I always got the message: 'Terminated' which made ACC unresponsive.
The same happens on the tablet (gta4xlwifi) with small differences.

I also noticed that in the latest LOS version there is a battery configuration option called 'Charging control' which is disabled by default. Can it interfere with ACC if it's disabled?

What could cause this behavior? Where can I find informative logs/infos about this 'Terminated' message?

I did not try to reinstall a previous version of ACC, I'm focusing on fixing the problem with the latest version.

tamas646, 2023-07-26:
**Additional informations:**
- There's a chance that the issue was already present in **v2023.6.26** and I did not notice it (because I was only using this version for a few days)
- Previously I did not have to change the charging switches. The automatically found ones was working quite well without any problem
- The issue can be also caused by a LOS upgrade. It is difficult to say...

VR-25, 2023-07-26:
1. Do you remember the exact last working version?
2. What about the last working charging switch? I know it was “automatic”, but in the config, one could see a charging switch set. Have you seen it?
3. "Terminated" refers to the daemon. It's shown if the daemon takes longer than 10 seconds to stop. Any command that stop/restart the daemon may print that message at some point, especially when the charger is unplugged. The 10 seconds timeout is for accd to restore all default charging settings before stopping.
4. LOS' _Charging control_ does some of what acc does. Having it disabled prevents conflicts.
5. Indeed, a LOS update may have triggered the issue.
6. You can revert to the previous installation and config, by running `acc -b`. This is for the future, since you've already installed more than one update.

VR-25, 2023-07-26:
While we're at it, here's a new dev build: 
[acc_v2023.7.26-dev_202307260_1900.zip](https://github.com/VR-25/acc/files/12175989/acc_v2023.7.26-dev_202307260_1900.zip)

tamas646, 2023-07-27:
1. I remember that the previously manually installed version was **v2022.6.4**. If the pre-release versions was available through the Magisk manager I'm surely have upgraded to it. So in this case the last working version was **v2022.7.30-dev**.
2. The last working charging switch for *evert* was `battery/battery_charging_enabled 1 0`, and for *gta4xlwifi* it was `battery/batt_slate_mode 0 1`
3. Okay. So it means that ACC failed to restore the default charging switches, right? Is there any log file where we can see exactly what have failed?

Thanks for the answers. I will try the dev build you sent.

tamas646, 2023-07-27:
**Update:**

On device *gta4xlwifi* after a reboot the version **v2023.7.17** now seems to be working. The 'Terminated' message has gone, the switches are working fine (also the `battery/batt_slate_mode 0 1`). Not sure whether it's only temporary or not. I'll give it some time.

As I experienced yesterday (before I submitted this issue) just only rebooting any of the devices did not solve the problem. I mean some part of ACC worked fine again, but a bit lately the 'Terminated' message showed up again.

On device *evert* I upgraded the ACC version to **v2023.7.26-dev** and rebooted:
1. `acc -t` ran fine without any error and the result seemed to be correct. But right after the `acc -D` showed that the daemon is not running, so I started it with `accd -i` (no error message)
2. Running `acc -t` again gave message `Terminated`
3. After a reboot `acc -t` also gave message 'Terminated'
4. Unplugged the charger but the phone still thought that it's charging (I remember that sometimes earlier it took long until it detected the unplug, sometimes it was few minutes)
5. After a minute it still thought that the charger was plugged in, so I powered off the device.
6. A few minutes later I switched on the phone and plugged in the charger again
7. `acc -t` ran fine without any error, `acc -D` showed that the daemon is running
8. Now ACC seems to be working...

Unfortunately I cannot reproduce the issue. I'm not sure what's causing it. All I know is sometimes it comes up, and switching off, unplugging the charger, waiting, switching on and re-plugging the charger solves the problem in most cases.

VR-25, 2023-07-27:
> Okay. So it means that ACC failed to restore the default charging switches, right? Is there any log file where we can see exactly what have failed?

No, 10 seconds timeout is more than enough to revert all changes. Acc just failed to reenable charging within that timeframe. However, the switch was reset successfully. The thing is, acc waits for charging to resume -- and if the charger is unplugged or the switch takes longer to respond, then you see the "Terminated" message, meaning, acc is killed to stop waiting.

Regarding the charging tests/control inconsistencies, see what the value of discharge_polarity is (acc -sp pol). If it's wrong, you can correct it with `acc -s dp=-` (or +). Acc determines the polarity only once, and when the charger is unplugged. This automatic detection may be wrong on devices with troublesome charging switches.

Speaking of switches, set the ones (with acc -ss) that you find working on both devices. acc -ss essentially disables the automatic switch testing. So that acc uses only what is already set.

closed 2023-11-30 by VR-25