VR-25/acc #206: MTK reboots on a switch and autoblacklisting doesn't work

usr97629238, 2023-08-10:
The issue seems similar to https://github.com/VR-25/acc/issues/107
Version is 2023.7.17
Killed accd, tried `acc -x -t`:
```
1/8: /proc/mtk_battery_cmd/current_cmd 0::0 0::1
...blah...
Switch works
- battIdleMode=false

2/8: /proc/mtk_battery_cmd/current_cmd 0::0 0::1 /proc/mtk_battery_cmd/en_power_path 1 0
...blah...
Switch works
- battIdleMode=false

3/8: /sys/devices/platform/battery_meter/FG_suspend_current_threshold 100 5
...blah...
Switch doesn't work

REBOOT
```
And the `/proc/mtk_battery_cmd/current_cmd` switch indeed works, tested manually with `acc -x -e / acc -x -d`
And it keeps trying that `/sys/devices/platform/battery_meter/FG_suspend_current_threshold` switch after the reboot, crashing again and again despite "_The upcoming release will auto detect and blacklist unwritable control files and those that trigger unexpected reboot_" and the `/data/adb/vr25/acc-data/logs/ps-blacklist.log` is empty.
Tried adding `/sys/devices/platform/battery_meter/FG_suspend_current_threshold` to the `/data/adb/vr25/acc-data/logs/ps-blacklist.log` file manually, tried commenting out that switch in the `/data/adb/modules/acc/ctrl-files.sh`, to no avail.

VR-25, 2023-08-15:
Auto-blacklisting does not account for delayed unexpected reboots, which is what seems to be happening.
If FG_suspend_current_threshold was manually blacklisted, and yet, you're still facing reboots, then it's not the culprit.

What's the full list of available switches? (Run `acc -ss:`)

VR-25, 2023-08-18:
Have you tried manually toggling `/proc/mtk_battery_cmd/en_power_path`?

closed 2023-11-30 by VR-25