VR-25/acc #214: Magisk default config for ACC/AccA

eugenesan, 2023-10-17:
**This not a bug but rather a question and maybe a request document the situation for new users.**

I just did clean device install with AccA and default Magisk settings ((Zygisk: disabled, MountNamespaceMode: Inherit)) and it didn't work properly.
The daemon appeared to be running bu AccA could not access acc command line (acc: inaccessible).

After altering Magisk settings ((Zygisk: Enabled, MountNamespaceMode: Global) and reboot issues solved.

VR-25, 2023-10-17:
Hm, that's weird...

Even if the system lacks writable /sbin/ to have acc executables in $PATH right after installation, AccA should still work, since it always uses the /dev/acc workaround (all executables linked to /dev/).

Are you able to run /dev/acc under the same conditions (clean device install with AccA and default Magisk settings)?

eugenesan, 2023-10-18:
I'll have to check that later.
At least for now if somebody will have similar issue they will find this issue and try my solution.

VR-25, 2023-11-30:
Looks like it had to do with 509eafa951a9fff33da3c72d1a306ae14ad82bd7 .

closed 2023-11-30 by VR-25