VR-25/acc #215: Charge/Discharge Bug

neoteroinvin, 2023-10-20:
Similar to #196 , I seem to have the problem where when the daemon is running, it will switch between charging and discharging. Stopping the daemon halts this behavior. I have it set to charge until 80% then discharge until 70%, then resume. A cooldown at max 60C down to 40C, and an applyonplug of wireless/voltage_max:9000000. I've attached logs from when this issue occurs. I'm on 202310160 using ACCA to change settings. I turned off prioritize battery idle mode to see if that had an effect and it did not. I noticed this change after upgrading via the magisk app from 202308190.
[acc-logs-barbet.zip](https://github.com/VR-25/acc/files/13048840/acc-logs-barbet.zip)

VR-25, 2023-10-21:
Have you tried another charging switch?
Unreliable switches tend to trigger unexpected behaviors.

neoteroinvin, 2023-10-21:
I have it set to automatically cycle through switches, I don't remember what switch it was set to when the behavior was being observed. A quick look at the logs I didn't find it either. It's currently not displaying the behavior with the sm7250_bms/charge_disable 0 1 switch though.

closed 2023-11-30 by VR-25