VR-25/acc #216: Wrong charging status on Nothing Phone 2, exit code 7

lehmartin, 2023-10-20:
Device: Nothing Phone (2) (PongEEU)
OS: Android 13/Nothing OS 2.0.3 (Pong-T2.0-230906-1933-EEA)

I have a weird behaviour on Nothing Phone 2: the shown status is the opposite the real status. When charging, ACCA says it is being discharged, and when unplugged it says the phone is being charged. Consequently, the charging switch does not work as intended, although the switch/file itself should be correct since manually setting a current limit works. But I guess ACC is being confused by the reported status. Can provide more details if needed! 

Attached are:
- output of `acc --info`
[acc-logs-Pong.zip](https://github.com/VR-25/acc/files/13055489/acc-logs-Pong.zip)
- output of `acc -le`
[acc_info.txt](https://github.com/VR-25/acc/files/13055490/acc_info.txt)
- Screenshots of AccA:

[image]

VR-25, 2023-10-20:
Did you get a notification to set discharge polarity?
What did you do?

If you didn't get the notification, see if you do after running `acc -s dp=`.

lehmartin, 2023-10-20:
Yes, I recall such a notification. I think plugging/unplugging the phone removed it? Did not think too much about it... Guess that's where I screwed up. Anyway, using the mentioned command, I got the correct charging polarity, it seems to work now. My bad, should have read the documentation more thoroughly. Thanks for your help, feel free to delete this issue!

closed 2023-10-20 by lehmartin

VR-25, 2023-10-21:
No worries.

When discharge polarity is not set, accd asks you to unplug for it to determine and set the correct value. You can also do it manually by running `acc -s dp=SIGN`