VR-25/acc #219: Manual idle mode toggle

scarlettekk, 2023-11-23:
Similar to the "charge once to #%" option, it would be cool to have an option to either enable idle mode until it is turned off, or enable idle mode for the current charging session, i.e. regardless of the current battery level the phone will just directly run off the charger and not charge the battery. This is useful for something like a midday car trip where the phone will be plugged in for an extended period of time but maybe shouldn't be charged for the sake of battery lifespan

VR-25, 2023-11-23:
Idle mode is enabled whenever charging is paused.
That said, there's a command for disabling charging on demand (`acc -d`).
To restore your regular settings, run `accd`.

From the documentation:
```-d|--disable [#%, #s, #m, #h or #mv (optional)]   Disable charging
    e.g.,
      acc -d 70% (do not recharge until capacity <= 70%)
      acc -d 1h (do not recharge until 1 hour has passed)
      acc -d 4000mv (do not recharge until battery voltage <= 4000mV)```

scarlettekk, 2023-11-23:
What happens if you `acc -d 1h`, wait an hour, and don't run `accd`?

VR-25, 2023-11-24:
Charging will not resume unless you reboot or run a command to enable it.

closed 2023-11-29 by VR-25