VR-25/acc #223: Install hangs on install/uninstall.sh flock

scarlettekk, 2023-11-26:
Trying to install this module on a Pixel Watch 2, and the install hangs. Checked logs, and it's stuck on `install/uninstall.sh install`. I added logging to uninstall.sh to see what that was doing, and it looks like it gets stuck running a `flock 0` that never returns. The log from `uninstall.sh` is as follows:
```
21: bin_dir=/data/adb/vr25/bin
22: busybox_dir=/dev/.vr25/busybox
23: magisk_busybox='/data/adb/ksu/bin/busybox /data/adb/magisk/busybox'
24: '[' -x /dev/.vr25/busybox/ls ']'
43: unset f bin_dir busybox_dir magisk_busybox
46: 2>/dev/null
46: exec
49: mkdir -p /dev/.vr25/acc
49: <>/dev/.vr25/acc/acc.lock
50: flock -n 0
51: read pid
52: kill 4628
53: timeout 10 flock 0
54: >/dev/null
54: 2>&1
54: kill -KILL 4628
55: flock 0```

VR-25, 2023-11-29:
[acc_v2023.11.26.1-dev_202311261_2133.zip](https://github.com/VR-25/acc/files/13503603/acc_v2023.11.26.1-dev_202311261_2133.zip)
Try that build.

scarlettekk, 2023-12-03:
That seems to have resolved the issue. I will add more details if I encounter any problems.

closed 2023-12-31 by VR-25