VR-25/acc #227: Install then unroot?

silajim, 2023-12-03:
I have rooted the phone, but due to the root shenanigans these latest days I want to unroot the phone but keep this functionality, can I do it? Is it even possible?

IslamAlOrabI, 2023-12-08:
Instead of that you can use a custom rom that has a **charging control** feature. like : **crdroid** - I think all roms have that feature

[image]

silajim, 2023-12-08:
I am not using a custom rom, so I don't have that feature, or access to it

VR-25, 2023-12-31:
This can be achieved with a kernel/ramdisk modified to run acc initialization script (`/data/adb/vr25/acc/service.sh`) on boot.
Ping a kernel dev.

closed 2023-12-31 by VR-25