VR-25/acc #228: battery charges for a brief moment in idle mode

laur89, 2023-12-04:
Using the only switch supporting `idle` mode: `battery/charge_control_limit 0 battery/charge_control_limit_max`

`--info` reports `STATUS=Idle` confirming idle mode works. However if I switch the screen of and turn it back on, then `--info` displays `STATUS=Charging` for couple of seconds. This means it continues charging for some short period at this stage, i.e. turning screen off&on causes the battery level to creep up gradually. Is this expected?

Note all the config is default besides setting the switch.

Xiaomi 9T Pro (raphael), ArrowOS 11 (based on Android 11)

VR-25, 2023-12-31:
This is usually caused by bad thermal management. Try the latest development build.
[acc_v2023.12.25-dev_202312250_1039.zip](https://github.com/VR-25/acc/files/13800519/acc_v2023.12.25-dev_202312250_1039.zip)

laur89, 2024-01-01:
I'll have to close this issue for the time being. Xiaomi 9T pro I was testing this on got bricked. If I manage to restore it, I'll give it another go.

closed 2024-01-01 by laur89