VR-25/acc #233: FR: continue respecting rc when rt has value w/ r-postifx

laur89, 2024-01-10:
This ticket is to contain the work/discussion stemming from a [telegram discussion](https://t.me/acc_group/82451).

**As-is:**
If both `rc` (resume_capacity) and `rt` (resume_temp)  with the `r` postfix are defined, then configured rc value [becomes essentially no-op](https://t.me/acc_group/82382) by temporarily defining rc to pc-1. This means charging will continue regardless of the value we set for rc in config.

**To-be:**
Change the logic so that original/set value of `rc` continues to be respected. This means when cooldown has not been triggered, then battery should be allowed to discharge until  the original `rc` value. 
Am not familiar with the implementation, so can't tell whether the `pc-1` should be set only when cooldown kicks in _during_ charging, or the whole logic requires a rework.

**Reasoning:**
All this is to allow the battery to be discharged to our set `rc` value in non-idle mode. As it stands currently, battery would kept charged to `pc`.

VR-25, 2024-01-13:
Without the trailing `r`, acc requires two conditions to resume charging:
1. Battery capacity <= rc
2. Battery temp <= rt

With the trailing `r`, acc resumes charging when battery temp <= rt, regardless of rc.
Additionally, if charging has been paused by cooldown, it resumes regadless of rt and rc, when the cooldown conditions allow it.

The suggested change effectively kills the need for `r`:

> ...only in case a charging cycle has been stopped due to reaching a mt-condition (maybe also if paused by the cp-phase of a cch/cp cycle) rc is changed to pc-1 but then - as soon as charging was restarted (by reaching rt or the cch-phase of a cch/cp cycle) rc as defined applies again.

VR-25, 2024-01-26:
This change was implemented.

closed 2024-01-26 by VR-25