VR-25/acc #234: Daemon stopped with exit code 7 on a Fresh Install of Xiaomi HyperOS

ashwinprakas, 2024-01-17:
Daemon stops with exit code 7 on a Fresh Install of Xiaomi HyperOS(A14), was working perfectly fine on the MIUI(A13) and AOSP(A14) roms.
I tried to generate logs but didn't get a response.

VR-25, 2024-01-17:
Try pulling the latest snapshot by running `acc -u -f dev`.
If you stumble upon a curl error...

https://github.com/Zackptg5/Cross-Compiled-Binaries-Android/raw/master/curl/curl-arm64

Download that binary, rename to curl, create the directory /data/adb/vr25/bin/, place the file there and set permissions to 755.

ashwinprakas, 2024-01-18:
Error on my part.

'Nighttime Charging Protection' was ON and it was causing the issue, after disabling it, ACC Settings works as expected.

closed 2024-01-18 by ashwinprakas