VR-25/acc #236: Acc reporting non-sensical current, and multiple switches stopped working

WyntrHeart, 2024-01-26:
A few weeks ago acc was working correctly and I was using the switch "battery/charging_enabled" which worked and allowed idle mode. I remember testing all switches and seeing at least two with idle mode working.

Some time in the last few weeks this just stopped working, without me making any changes to my phone, acc, or anything in Termux. Now when I check the battery status info or test switches all of the ampere measurements are wildly inflated, showing charge *and* discharge rates over 4 amps, despite the phone appearing to actually charge and discharge at its normal rate. And the switch I was previously using with idle mode is no longer working *at all*, and of the three switches that still work none of them support idle mode, which is an essential feature for me.

I don't even know where to start to diagnose this, any ideas?

VR-25, 2024-01-26:
Upgrade to the latest build and see whether the issue persists.
You can do that directly with `acc -u dev` or by flashing the zip:
[acc_v2024.1.26-rc_202401260.zip](https://github.com/VR-25/acc/files/14069231/acc_v2024.1.26-rc_202401260.zip)

WyntrHeart, 2024-01-26:
I already tried "acc -u". Adding "dev" to the command gets me the following output:
`curl: option --dns-servers: the installed libcurl version doesn't support this
curl: try 'curl --help' or 'curl --manual' for more information

No update available`

VR-25, 2024-01-26:
Flash the zip.

WyntrHeart, 2024-01-26:
The zip installed without any issues. The amp measurements are fixed, but I haven't had time to test the switches yet. I'll let you know when I do

WyntrHeart, 2024-01-28:
The zip installed without issue, and fixed both the amp readings and the switches. Thank you for the help!

Any idea what I might have messed up to cause this issue to appear?

VR-25, 2024-01-28:
Could have been an issue related to the build you were running.

closed 2024-03-07 by VR-25