VR-25/acc #237: Battery jump between idle and charging during idle mode

feather-47, 2024-01-27:
Phone: Galaxy Note 10
ROM: SealROMv7
I've flashed the newest .zip (202401260), `acc -i` report `STATUS=Idle` with `charging_switch=battery/hmt_ta_charge 1 0 --` so Idle mode work. When running an app in `idle_app`, i noticed that battery slowly charged up, `acc -w` showed the battery to be idle for ~11s then switched to charging for also ~11s then repeat. I've only tried to address it with `accd` and rerunning the test with `acc -t` and `acc -ss`.

`acc -le`
[acc-logs-d1.tgz.zip](https://github.com/VR-25/acc/files/14072682/acc-logs-d1.tgz.zip)

VR-25, 2024-01-28:
The Android API used for getting information of the currently running foreground app is not very reliable. I'll add a tolerance to mitigate the fluctuations.

axet, 2024-02-01:
I have similar issues with my Nexus 5 device. When I ran 'acc 50 40' device is not charging. But monitoring CURRENT= showing values from -2000 to 2000 it seems charger is not turned off completely and keep maintaining battery level.

Web showing some answers:

* https://android.stackexchange.com/questions/200032/difference-between-system-files-charging-enabled-vs-battery-charging-enabled

Here are two modes available for devices: 'ac/charging_enabled' and 'battery/battery_charging_enabled' first suppose to cut power off completely and battery_charging_enabled suppose to maintain current battery charge level. Seems like some devices misbehave and using only one preferred mode. It maybe kernel issue or IC issue.

feather-47, 2024-02-26:
Idle mode worked properly with new dev update

closed 2024-02-26 by feather-47