VR-25/acc #241: Fast charging & Mar 2024 update

suddenlydoomy, 2024-03-07:
Fast charging doesn't seem to work after installing the Mar 2024 Android update for v2023.10.16.

Disabling the ACC module in Magisk and rebooting enables fast charging.

VR-25, 2024-03-07:
The latest version is 2024.2.25-rc. Is that what you're running?

suddenlydoomy, 2024-03-07:
If this is correct, then I sincerely apologize for wasting your time.  But GitHub says that the latest version is v2023.10.16.

Where is this release candidate?

VR-25, 2024-03-07:
[acc_v2024.2.25-rc_202402250_1841.zip](https://github.com/VR-25/acc/files/14530877/acc_v2024.2.25-rc_202402250_1841.zip)

Usually, those builds are published only on telegram.
Anyone can upgrade to the latest development build by running `acc -u dev`.

suddenlydoomy, 2024-03-08:
Still experiencing the same issue with that build.

VR-25, 2024-03-08:
Share your config (`acca - sp`).

closed 2024-03-08 by suddenlydoomy