VR-25/acc #242: Pixel 5 Idle Configuration Help

skream27, 2024-03-07:
Hi there, I'm using Google Pixel 5 but I don't understand anything from the ACC configuration. I want it to stop when it reaches 80% and switch to "battery idle" mode, that is, I want it to be powered from the adapter, not from the battery. I tried to do it using AccA, but I see that it consumes power from the battery even though the "battery idle" mode is on. Can anyone help me for create configuration and install the configuration?
Thank you so much.

VR-25, 2024-03-07:
Yoi don't have to create anything, let alone edit the config directly.

Simply install the [latest](https://github.com/VR-25/acc/files/14530877/acc_v2024.2.25-rc_202402250_1841.zip) build and run `acc 80`.

skream27, 2024-03-08:
> You don't have to create anything, let alone edit the config directly.
> 
> Simply install the [latest](https://github.com/VR-25/acc/files/14530877/acc_v2024.2.25-rc_202402250_1841.zip) build and run `acc 80`. That's all.

it will stop at 80 and use the USB power, am i wrong? i will try this night

VR-25, 2024-03-30:
If you're still facing issues with this, install the latest build.
You can upgrade directly with `acc -u dev`.
Development flashable zips are provided at https://t.me/acc_group .

closed 2024-03-30 by VR-25

VR-25, 2024-04-07:
[acc_v2024.4.7.1-rc_202404071_1214.zip](https://github.com/VR-25/acc/files/14896991/acc_v2024.4.7.1-rc_202404071_1214.zip)