VR-25/acc #243: Redmi 7 issues

hiddener, 2024-03-11:
Hello there! Running latest lineageos 20 and acc v2024.2.25-rc dev builds on onclite. I've manually tested the 26 switches listed with `acc -ss`. I've noticed the following:

- `battery/input_suspend 0 1 --` the only switch that has a charging sound and icon when plugging the cable. However, `acc --info` says it's idle, so it's not charging. I changed the thresholds several times.

- `battery/battery_charging_enabled 1 0 --` the only switch `acc -t` marked as works, but after selecting it, it didn't charge too.

- `battery/charging_enabled 1 0 --` status changed from idle to discharging, re-plugging the cable did nothing.

The other switches didn't work. Has anyone a working acc on onclite? Any suggestion highly appreciated!

VR-25, 2024-03-17:
Try resetting your config with `acc -sr`.

AnthonyTongLee, 2024-03-20:
I've an onclite as a secondary phone I've recently put lineageos on. Same issues in that charging works for a few seconds with `battery/battery_charging_enabled 1 0 --` then stops charging as going by the LED indicator. The battery indicator icon in the status bar is charging though.

As below I noticed in the battery info that the max was zero, I don't know if this means it wasn't set or is actually zero, but configuring `maxChargingCurrent=(1000)` seems to make it stay charging. 🤷‍♂️
```
  Max charging current: 0
  Max charging voltage: 0
```

VR-25, 2024-03-30:
Investigating ROM/kernel side issues.
Stay tuned for the next builds.

VR-25, 2024-04-07:
[Uploading acc_v2024.4.7.1-rc_202404071_1214.zip…]()

VR-25, 2024-04-07:
[acc_v2024.4.7.1-rc_202404071_1214.zip](https://github.com/VR-25/acc/files/14897002/acc_v2024.4.7.1-rc_202404071_1214.zip)

VR-25, 2024-04-11:
https://github.com/VR-25/acc/files/14951962/acc_v2024.4.11-rc_202404110_2144.zip

VR-25, 2024-04-17:
https://github.com/VR-25/acc/files/15017403/acc_v2024.4.17-rc_202404170_2102.zip

Install and reset config with `acc -sr`. If issues persist, contact kernel/ROM developers.

closed 2024-04-17 by VR-25