VR-25/acc #245: Charging Does Not Stop At Pause Capacity

Abeefe, 2024-03-24:
ACC does not stop charging at pause capacity no matter what version I use whether it is stable, release candidates, Dev version or the one you send in a personal chat with me.

I do have a working idle mode and it use to work before.

VR-25, 2024-03-30:
Message me on TG.

closed 2024-03-30 by VR-25

Abeefe, 2024-03-31:
TG has banned all my numbers, I was using a friend phone to chat with you before but he is out of town.

VR-25, 2024-03-31:
Try this build: https://github.com/VR-25/acc/files/14812954/acc_v2024.3.30-rc_202403300_1924.zip
Run `acc -s dp=` after installing.

VR-25, 2024-04-07:
[acc_v2024.4.7.1-rc_202404071_1214.zip](https://github.com/VR-25/acc/files/14896990/acc_v2024.4.7.1-rc_202404071_1214.zip)

Abeefe, 2024-04-07:
I was having the issue with the previous version, i think maybe i have faulty pmic and i want change it.

I will try this out too.

Abeefe, 2024-04-16:
During the day phone charges to100% without respecting the pause capacity, while at night profile pause capacity is respected.

This happens both on Stable and R.C.

VR-25, 2024-04-16:
This is a device issue.
Refer to readme > troubleshooting > charging switch.

If the build below has the same behavior, consider trying another charging switch - and ensure your ROM's own battery care features are all turned off.

https://github.com/VR-25/acc/files/15000886/acc_v2024.4.15-rc_202404150_1809.zip

Abeefe, 2024-04-17:
On Termux "su acc" is returning "segmentation fault.

VR-25, 2024-04-17:
> On Termux "su acc" is returning "segmentation fault.

[acc_v2024.4.17-rc_202404170_2102.zip](https://github.com/VR-25/acc/files/15017403/acc_v2024.4.17-rc_202404170_2102.zip)