VR-25/acc #248: v2024.4.7-rc fails with segfault

elhennig, 2024-04-07:
I tried to update acc via 

acc -u dev

but in version v2024.4.7-rc the daemon cannot be started and the acc command fails with segfault. I reverted back to the latest stable at that works fine.
# acc -v                                                                                                                        
v2023.10.16 (202310160)

elhennig, 2024-04-07:
Thanks for the clarification. Strange is, that it works with the latest stable release, but not with the dev version.

VR-25, 2024-04-07:
[acc_v2024.4.7.2-rc_202404072_1819.zip](https://github.com/VR-25/acc/files/14898136/acc_v2024.4.7.2-rc_202404072_1819.zip)
Try this.

elhennig, 2024-04-08:
No change, unfortunately. And I have no real idea how to debug this more to find possible issues on my device.

VR-25, 2024-04-09:
See if pasting and running the following line on terminal helps:

`cd /data/adb/vr25; mkdir -p bin; echo -e "#!/system/bin/sh\n:" > bin/pkill; chmod 0755 bin/pkill`

VR-25, 2024-04-11:
[acc_v2024.4.11-rc_202404110_2144.zip](https://github.com/VR-25/acc/files/14951962/acc_v2024.4.11-rc_202404110_2144.zip)

xChickens, 2024-04-13:
> [acc_v2024.4.11-rc_202404110_2144.zip](https://github.com/VR-25/acc/files/14951962/acc_v2024.4.11-rc_202404110_2144.zip)

Yea I'm also getting segfault on this release and the latest weekly release. On Pixel 7 Pro latest rom with APatch

kailiu42, 2024-04-15:
I'm from issue #249. It seems this segfault is due to `data/adb/vr25/acc/acc.sh`:

```
#!/system/bin/sh
#exec_wrapper
if [ -f /dev/.vr25/acc/.updated ]; then  <-- no such .updated file, so the script goes to the else branch
  exec /dev/accd. "$@"
else
  exec . /data/adb/vr25/acc/acc.sh "$@"  <-- runs into a recursive loop and sh finally segfault.
fi
```

VR-25, 2024-04-16:
> I'm from issue #249. It seems this segfault is due to `/data/adb/vr25/acc/acc.sh`:
>
> ```
> #!/system/bin/sh
> #exec_wrapper
> if [ -f /dev/.vr25/acc/.updated ]; then  <-- no such .updated file, so the [...]

That file is in the bin folder. It's not acc.sh.

Are all of you using Apatch? Why am I not getting segfaults with KernelSU?

kailiu42, 2024-04-16:
> That file is in the bin folder. It's not acc.sh. So, no chance of recursive loop there.

This file is `/data/adb/vr25/acc/acc.sh`, that's why it's segfaulting.

I'm using Magisk. Your magisk module might have some problem.

kailiu42, 2024-04-16:
This is how it looks like after flashed acc_v2024.4.11-rc_202404110_2144.zip.

[image]

VR-25, 2024-04-16:
> This is how it looks like after flashed acc_v2024.4.11-rc_202404110_2144.zip.
> 
> <img alt="Snipaste-20240416-160432" width="823" src="https://private-user-images.githubusercontent.com/358679/322757820-d2f7e9ba-d885-4e2f-8adf-20c197cc0026.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMyNTU5NDgsIm5iZiI6MTcxMzI1NTY0OCwicGF0aCI6Ii8zNTg2NzkvMzIyNzU3ODIwLWQyZjdlOWJhLWQ4ODUtNGUyZi04YWRmLTIwYzE5N2NjMDAyNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQxNlQwODIwNDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02NGM0ODNmNDVmYjAxZTkxZGQ1OThmMWVhNTU5NTcwMjczZTlhMWVlY2RlZDE0Mjc0ZGY1NzE0NDNhMGNlYWM3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.nI_Iftk97E-5Rk_Yh_WsfswEAodhVdjMcDVJDc9WPgo">

Definitely something's wrong there.
acc.sh in this case, has the contents of bin/acc, which is weird.
Looks like this is not happening to KernelSU Installs.
I'll take a closer look.

VR-25, 2024-04-16:
I'd like you guys to try a clean install.
Uninstall acc with `acc -U` then flash the new zip.

[acc_v2024.4.15-rc_202404150_1809.zip](https://github.com/VR-25/acc/files/15000886/acc_v2024.4.15-rc_202404150_1809.zip)

Even though acc uninstalls the current version before an upgrade/downgrade, there's an issue when upgrading from a fairly older build. It mistakenly follows bin/acc* symbolic links, overwriting the wrong files. I'll be working on a fix soon.

kailiu42, 2024-04-17:
> I'd like you guys to try a clean install. Uninstall acc with `acc -U` then flash the new zip.

Yes the segfault is gone now.

VR-25, 2024-04-17:
https://github.com/VR-25/acc/files/15017403/acc_v2024.4.17-rc_202404170_2102.zip

closed 2024-04-17 by VR-25