VR-25/acc #249: Charge % limit not working with wireless charging on OnePlus 12

kailiu42, 2024-04-11:
I have opened a issue at https://github.com/MatteCarra/AccA/issues/242 but later realized that here might be the right place.

I recently get a new OnePlus 12 and find out although charge % limiting works with wired charging, it just doesn't work with wireless charging at all.

Please let me know what information and logs you need and how to collect them. Thanks for the great tool btw.

VR-25, 2024-04-11:
https://github.com/VR-25/acc/files/14951962/acc_v2024.4.11-rc_202404110_2144.zip

Hi, try the latest build from the link above.
If `acc -t` still shows no positive results, share the output of `acc -p`.

Note that wireless charging is different. Oftentimes, having a switch working with wired charging doesn't necessarily mean it'll also work with other charging methods. This is a device's own limitation.

kailiu42, 2024-04-12:
After flashed this new build, I got Segmentation fault while trying to run `acc -t` or `acc -v`.

kailiu42, 2024-04-17:
Tried v2024.4.15, `acc -t` shows switch `/sys/devices/virtual/oplus_chg/battery/mmi_charging_enable` works when testing with wireless charging. But after test is done, wireless charging % limiting does not work.

VR-25, 2024-04-17:
New build: https://github.com/VR-25/acc/files/15017403/acc_v2024.4.17-rc_202404170_2102.zip

You have to find a suitable switch as described in readme > troubleshooting > charging switch.

Perhaps try `acca -t p` as well to test potential switches.

Some switches only work under specific conditions.
Issues with wireless charging are not new.

Also worth trying `force_off=true`.

closed 2024-04-17 by VR-25