VR-25/acc #250: Pixel 6 latest version intermittent charging

lavoie005, 2024-04-18:
hi,thank you for your time.
i use a pixel 6 something weird happen. Yesturday i was on android 13 and i have update it to android 14 today.
i have install your latest version but it intermittent working for limiting the voltage (4.06v) .
so i decide to donwgrade with your version v2022.6.4 old one install in my pixel 6 on android 13.
it now work like a charm.
any idea why your latest version cant limit the voltage power but your old one can?
have a good one and thanks again

VR-25, 2024-04-18:
https://github.com/VR-25/acc/files/15017403/acc_v2024.4.17-rc_202404170_2102.zip

VR-25, 2024-04-27:
https://github.com/VR-25/acc/files/15137601/acc_v2024.4.26-rc_202404260_0003.zip

VR-25, 2024-05-11:
[acc_v2024.5.11-rc_202405110_1223.zip](https://github.com/VR-25/acc/files/15282648/acc_v2024.5.11-rc_202405110_1223.zip)

closed 2024-05-11 by VR-25