VR-25/acc #251: damon stopped with exit code 1 pixel 8 A14

nutpantz, 2024-04-22:
stops every time in lees then two minutes
i assume the logs are all the same but here are 4

nutpantz, 2024-04-22:
[logs.zip](https://github.com/VR-25/acc/files/15057774/logs.zip)

VR-25, 2024-04-22:
Try the following build:
[acc_v2024.4.22-rc_202404220_2049.zip](https://github.com/VR-25/acc/files/15068041/acc_v2024.4.22-rc_202404220_2049.zip)

VR-25, 2024-04-23:
Nice.
Now, could you share a fresh log archive to see what's behind the incorrect reports (current, voltage, etc.)?

nutpantz, 2024-04-23:
[acc-logs-husky_2024-04-23_04:43:38.tgz](https://github.com/VR-25/acc/files/15076269/acc-logs-husky_2024-04-23_04.43.38.tgz)

These help?

VR-25, 2024-04-23:
Yes.
I'll take a look soon.

nutpantz, 2024-04-27:
Another fresh log with latest from my pixel 6
[acc-logs-oriole_2024-04-27_08:41:03.tgz](https://github.com/VR-25/acc/files/15138737/acc-logs-oriole_2024-04-27_08.41.03.tgz)

(Fit the lack of charging details)

nutpantz, 2024-04-27:
And another fresh from my pixel 8pro
[acc-logs-husky_2024-04-27_08:34:52.tgz](https://github.com/VR-25/acc/files/15138740/acc-logs-husky_2024-04-27_08.34.52.tgz)

Anything else i can do to help. Just let me know

VR-25, 2024-04-27:
It's now an AccA's own fault.
If you run acc -i, you'll see charging details.

Perhaps I'll think of workarounds for the outdated AccA.

nutpantz, 2024-04-27:
Is acca abandoned?
Should i make a bug issue there so people know?

VR-25, 2024-04-28:
The developer stopped supporting it over two years ago.

nutpantz, 2024-04-28:
I was hoping it was still supported.
I'll make a bug issue there
And hopefully someone interested will fork it

It's not a big deal for me, i just need the charging to stop and start correctly. 
(And a way to test settings)

Thanks for all your work.

VR-25, 2024-05-11:
Here's a new build: [acc_v2024.5.11-rc_202405110_1223.zip](https://github.com/VR-25/acc/files/15282648/acc_v2024.5.11-rc_202405110_1223.zip)

closed 2024-05-11 by VR-25